Opentrons/opentrons #8247: feat: custom calibration feature definition in labware definition file

radmanahmadi, 2021-08-19:
# Overview
The default calibration well is A1. There are a number of issue with this:
- It can be difficult for a user to visually calibrate to the center of a tube, especially when the tube top is large. 
- It becomes difficult to calibrate in the y-axis when the labware is near the back of the instrument or multiple opentrons are sitting side by side.
- Certain custom labware can have features that obscure the top of the calibration tube. 

It would be a lot easier if the scripter could define their own calibration location. They would choose a tube that is in a better position on the deck, or they would create their own calibration features on their custom labware. 

A potential workaround is to set the coordinates of A1 to the desired calibration location, and have a different well point to the physical location of A1. This method is very messy, and can potentially break code that iterates over all the wells on a labware. 

# Implementation details
The JSON parser should search the labware definition file for an object called "calibration_location". This object will have three sub-objects: x, y, z, and offset_z. Here the scripter can explicitly define the location that the opentrons will use to calibrate. 

"offset_z" will define a z offset that will be used during the first labware calibration. In some cases the scripter may choose to calibrate to a flat location (such as an etching or marking). To prevent the tip crashing into a surface, the instrument will send the pipette tip to z+offset_z. 

When "calibration_location" is not found, the default calibration location (A1) should be used. 

# Design

# Acceptance criteria
- "calibration_location", found in the labware definition file, will be used for labware calibration. 
- If "offset_z " is defined, the instrument will use this offset during the first calibration to avoid crashing the tip. 
- The software must default to A1 when there is no "calibration_location" found in the labware.

SyntaxColoring, 2021-08-19:
Thanks for the feature request!

I agree, this would be helpful.

One thing that makes this kind of nontrivial is what instructions would be shown to users in the Opentrons App. Currently, we have stock animations for different common labware types, and they all assume you're calibrating to well A1.

radmanahmadi, 2021-08-19:
There could be a fourth object in the labware file that defines a set of instructions for calibration. As for diagrams, I think it will need to be up to the scripter to provide more detailed instructions.