Opentrons/opentrons #8272: bug: Fast simulation uses APIv2.5 and before flow rates

alexjoel42, 2021-08-30:
# Overview

## Establishing background: 

In APIv2.6 we changed the [flow rates](https://github.com/Opentrons/opentrons/pull/6283) which is reflected in our [API documentation ](https://docs.opentrons.com/v2/new_pipette.html#defaults). In [4.5](https://github.com/Opentrons/opentrons/blob/edge/CHANGELOG.md#450-2021-08-03) we changed our simulation's default behavior to a [faster simulation](https://github.com/Opentrons/opentrons/pull/8022). 

## The bug 
Now when protocols are run on the Ot-2, regardless of API version they go to the APIv2.5 and before defaults. This causes unexpected default volume changes for customers. 

# Steps to reproduce

1. Write a protocol with a GEN2 Pipette
2. Have the OT-2 on 4.5 
2. Load the protocol with aspiration/dispense
3. The defaults will have changed

# Current behavior

1. Load protocol below with the faster simulation default
2. It will go to 137.35 uL/sec instead of 274.7 uL/sec

```python3
from opentrons import protocol_api, types

metadata = {'apiLevel': '2.11'}

def run(protocol: protocol_api.ProtocolContext):
    tr2 = protocol.load_labware('opentrons_96_filtertiprack_1000ul', '1')
    tr1 = protocol.load_labware('opentrons_96_tiprack_300ul', '2')
    p1000 =  protocol.load_instrument('p1000_single_gen2', 'right',tip_racks=[tr2])
    plate = protocol.load_labware('corning_96_wellplate_360ul_flat', '4')
    reservoir = protocol.load_labware('nest_12_reservoir_15ml', '6')
    p1000.pick_up_tip()
    p1000.aspirate(360, reservoir['A1'])
    p1000.dispense(360)
```

[image]

# Expected behavior

1. Should be the same as when simulated on opentrons_simulate 

```
Picking up tip from A1 of Opentrons 96 Filter Tip Rack 1000 µL on 1
Aspirating 360.0 uL from A1 of NEST 12 Well Reservoir 15 mL on 6 at 274.7 uL/sec
Dispensing 360.0 uL into A1 of NEST 12 Well Reservoir 15 mL on 6 at 274.7 uL/sec
```

# Work around
Turn on feature flag **Use Older Protocol Analysis Method**

mcous, 2021-08-31:
If the workaround for this issue is simply turning fast analysis off, then that means the bug is entirely contained to the run log print-out, and that **the actual flow rates used during run will be correct**. Simulation behavior has no bearing on actual execution behavior, and the run log **UI** is merely a (very) naive replaying of the simulation data, even if what the robot is actually doing during the run is different.

(Setting aside the incorrect flow rates being used during the analysis phase, the whole "run log is just a replay of the - potentially different - simulation results" is not a thing that we're happy about, and will not be the case in ProtocolEngine-based runs)

alexjoel42, 2021-08-31:
# This is my first attempt at solving a bug so apologies in advance 

My best guess is that something is happening in the [protocol_context.py](https://github.com/Opentrons/opentrons/blob/be4b1b507a637f71db8c2bb4cb95caf4dcbb086f/api/src/opentrons/protocol_api/protocol_context.py) `code in lines 258 to 266` 

So I think in the `InstrumentContextSimulation` might be using a different pipette default. If I were to guess it is likely within the four places that came up when I searched [InstrumentContextSimulation](https://github.com/search?q=org%3AOpentrons+InstrumentContextSimulation&type=code)

So then I think (because it is something wrong with the instrument) that the way that we are using `InstrumentContextSimulation` within the fast simulation is likely the culprit which is defined [here](https://github.com/Opentrons/opentrons/blob/d7e94ed42d9aa40e38fe584a08cc15443a896d84/api/src/opentrons/protocols/context/simulator/instrument_context.py#L18) 

So then investigating the InstrumentContextSimulation, I find that this function takes in `AbstractInstrument`. 

So I think that we then move to here with all of the  `AbstractInstrument` methods [here] (https://github.com/Opentrons/opentrons/blob/d7e94ed42d9aa40e38fe584a08cc15443a896d84/api/src/opentrons/protocols/context/instrument.py#L15) 

So if my logic is right then we have [util.py with how the flow rates are defined ](https://github.com/Opentrons/opentrons/blob/d7e94ed42d9aa40e38fe584a08cc15443a896d84/api/src/opentrons/protocols/api_support/util.py#L118) 

# Best effort to provide the beginning of resolution steps 

## We don't set the [API](https://github.com/Opentrons/opentrons/blob/d7e94ed42d9aa40e38fe584a08cc15443a896d84/api/src/opentrons/protocols/context/simulator/instrument_context.py#L33 ) for the default flow rates. So they default to the older API version. 

[image]

## This is in fact different from how we actually define i[nstrument_context.py](https://github.com/Opentrons/opentrons/blob/d7e94ed42d9aa40e38fe584a08cc15443a896d84/api/src/opentrons/protocols/context/protocol_api/instrument_context.py#L50)

[image]

# Here are the logs from my protocol which confirm your statement. 

 
[api.log](https://github.com/Opentrons/opentrons/files/7080931/api.log)
[serial.log](https://github.com/Opentrons/opentrons/files/7080936/serial.log)
[server.log](https://github.com/Opentrons/opentrons/files/7080940/server.log)

PR #8282 (2021-09-01): fix(fastsim): set_defaults for FlowRates in InstrumentContextSimulation
# Overview

Default flow rates are not correct when using fast simulation. 

closes #8272 

# Changelog

set default flow rates based on api_version in `InstrumentContextSimulation`

# Review requests

# Risk assessment

None
commit 165dd5d4 (2021-09-01): fix(fastsim): set_defaults for FlowRates in InstrumentContextSimulation

closes #8272
closed 2021-09-01 by amitlissack