Opentrons/opentrons #8294: feat: Let protocols define a no-fly zone

SyntaxColoring, 2021-09-02:
(Migrated from an internal wiki page)

# Use Case

Some users request never moving over a certain area on the deck due to contamination concerns.

# Current Workaround

Determining where on the deck the pipette is before a possible conflict of moving over the “no fly” zone, and hard-coding in an intermediate movement to ensure the zone is not interfered with.

SyntaxColoring, 2021-09-02:
Alternatively, I wonder if we could put this in the protocol author's control by defining a good set of safe points, like:

* The center of a deck slot (at some arbitrary safe height)
* Exactly in between two deck slots (at some arbitrary safe height)

And then the protocol author could plan the route themselves with a series of `move_to()`s.

```python
# Suppose this happens in slot 1.
pipette.aspirate(100, plate['A1'])
pipette.dispense(100, plate['B1')

# Now we want to drop the tip in the trash, but we want to avoid passing over slot 4.
pipette.move_to(center_of_slot(3))  # Move to the right.
pipette.drop_tip()  # Automatically takes the direct path to the trash bin,
                    # which now avoids slot 4 because we moved to the right beforehand.
```

mkeegan, 2021-09-07:
@SyntaxColoring this is exactly how we have solved the issue for our own system. We have a set of 'safe' points where we know we will not collide with our uncapping machine. We add fake microplates as labware and then use the well position to give us more granular positioning without modding the SDK.

[image]

closed 2025-12-17 by SyntaxColoring