Opentrons/opentrons #8295: feat: Tying a list of tips, rather than a list of tip racks, to a pipette

SyntaxColoring, 2021-09-02:
(Migrated from an internal wiki page by @ncdiehl11)

# Use Case

Users often want to iterate through tips in a non-standard way. An example is picking up fewer than 8 tips with a multi channel pipette— in this case, the user may want to iterate through row H of the rack, followed by row G, then row, F, etc.

# Current Workaround

Define a list of tips as mentioned above in the correct order of iteration ( [tip for row in tiprack.rows()[::-1] for tip in row] ). Then, create a custom pick_up function that increments a tip counter and iterates through this list of tips.

silvtal, 2022-09-30:
Isn't this done already?  [#11038](https://github.com/Opentrons/opentrons/pull/11038)

SyntaxColoring, 2022-09-30:
Nah, not quite. This issue is referring to how you can currently do this:

```python
pipette = protocol.load_instrument("p300_single_gen2", mount="left", tip_racks=[tip_rack_1, tip_rack_2])
# ...
# Automatically uses the next available tip in [tip_rack_1, tip_rack_2].
pipette.transfer(100, well_plate["A1"], well_plate["A2"])
```

But you can't do something like this:

```python
tips_to_use = [tip_rack_1["A1"], tip_rack_2["A1"], tip_rack_1["G6"], tip_rack_2["G6"]]
pipette = protocol.load_instrument("p300_single_gen2", mount="left", tips=tips_to_use)
# ...
# Automatically uses the next available tip in tips_to_use.
pipette.transfer(100, well_plate["A1"], well_plate["A2"])
```

In other words, this ticket is about implicit tip pickups.

#11038 is about adjusting the exact tip pickup position when you have an *explicit* tip pickup, like `pipette.pick_up_tip(tip_rack["A1"].top(-1))`.

SyntaxColoring, 2024-02-21:
Migrated to RSS-483.

closed 2024-02-21 by SyntaxColoring