Opentrons/opentrons #8315: bug: blow_out = False doesn't keep Distribute command from blowing out at trash

urihs, 2021-09-08:
# Overview

Use this section to describe your bug at a high level. Please include any issues you can find that may be related.

# Steps to reproduce

In version 4.5.0, 4.4.0 or 4.3.1, try to run the distribute function from a tube (tuberack: opentrons_24_tuberack_nest_1.5ml_snapcap) to a whole column in a wellplate (opentrons_96_tiprack_20ul). It's a p20 single GEN2 pipette distributing the max volume.

```
    p20s.distribute(20,                                             
                    tuberack.wells_by_name()["B6"],                           
                    plate.columns()[0],        
                    blow_out=False,  
                    touch_tip = True,                    
                    trash=True)          
```

# Current behavior

For each well, the pipette aspirates and dispenses twice. Each time, it aspirates 11 uL in order to be able to blow at the trash:
```
Distributing 20.0 from B6 of Opentrons 24 Tube Rack with NEST 1.5 mL Snapcap on 1 to A1 of NEST 96 Well Plate 200 µL Flat on 5
	Transferring 20.0 from B6 of Opentrons 24 Tube Rack with NEST 1.5 mL Snapcap on 1 to A1 of NEST 96 Well Plate 200 µL Flat on 5
		Picking up tip from A1 of Opentrons 96 Tip Rack 20 µL on 8
		Aspirating 11.0 uL from B6 of Opentrons 24 Tube Rack with NEST 1.5 mL Snapcap on 1 at 7.56 uL/sec
		Touching tip
		Dispensing 10.0 uL into A1 of NEST 96 Well Plate 200 µL Flat on 5 at 7.56 uL/sec
		Touching tip
		Blowing out at A1 of Opentrons Fixed Trash on 12
		Aspirating 11.0 uL from B6 of Opentrons 24 Tube Rack with NEST 1.5 mL Snapcap on 1 at 7.56 uL/sec
		Touching tip
		Dispensing 10.0 uL into A1 of NEST 96 Well Plate 200 µL Flat on 5 at 7.56 uL/sec
		Touching tip
		Blowing out at A1 of Opentrons Fixed Trash on 12
```
The workaround is setting disposal_volume to 0.

# Expected behavior

The pipette should take 20 uL and dispense only once for each well if the user sets blow_out to False.

mcous, 2021-09-16:
Ok, I see what's happening here. You're using `distribute`, and `distribute` happens to break the transfer down into a series of single pipetting actions due to the volume in play. This is certainly unintuitive, and I think related to #3117.

That being said, according to our [Complex Commands documentation](https://docs.opentrons.com/v2/new_complex_commands.html), `distribute` is primarily geared towards transfers that look like:

1. Aspirate liquid from well 0
2. Dispense in well 1
3. Dispense in well 2
4. Dispense in well 3

>  InstrumentContext.distribute() is optimized for taking a large volume from a single (or a small number) of source wells, and distributing it to many smaller volumes in destination wells. Rather than using one-to-one transfers, it dispense many times for each aspirate.

The `disposal_volume` is really important to ensure accurate dispenses from the initial, single aspiration. My apologies if I'm misreading your description, but it seems like your desired flow, instead, looks like:

1. Aspirate from well 0
2. Dispense in well 1
3. Aspirate from well 0
4. Dispense in well 2
5. Aspirate from well 0
6. Dispense in well 3

If that's the case, you should be using a [One-to-many `transfer` ](https://docs.opentrons.com/v2/new_complex_commands.html#one-to-many), instead. The `disposal_volume` is only applicable to `distribute`, so I think you'll avoid the disposal blow out in this case

silvtal, 2021-09-17:
Thank you for your response. I still don't understand why there's a disposal volume, but I will use loops of `transfer` calls for this kind of step then, as recommended.