Opentrons/opentrons #8349: feat: add 'attach CSV' to the protocol Open button in the OT App

splaisan, 2021-09-16:
It seems that when using CSV data to parametrize a protocol (not coming from the library), the CSV content needs to be injected in the top part of the protocol script.

Although this is quite straightforward for a programmer, it is not at all practical for lab users with no experience in computing and error prone.

Would it be possible to refer to the CSV file in the code and attach a CSV file during Open at runtime so that the OT App itself injects the CSV data in the dedicated string and further processes it (as already done now with the embedded data objects). Of course it will be the responsibility of the user to match the CSV to the expected format/columns

Adding an attach button to the open button or a drag and drop zone should fix this quite easily (for the wet lap user).

Thanks in advance,

Stephane

PS: Working with Jupyter on the OT2 PC to do this is NOT an option for wet-lab people.

mcous, 2021-09-16:
Thanks for the feedback @splaisan! We agree that this is an annoying pain point for app usage. We've been running a [beta test of a new HTTP protocol upload API](https://github.com/Opentrons/http-api-beta), which allows you to upload data files (like CSVs) alongside the protocol itself.

We are actively working on overhauling the protocol upload UI in the app, and we will be using this newer, multi-file-compatible API to back the app. We'll update this thread when we have a more accurate timeline for the release of this feature

splaisan, 2021-09-16:
Thanks Mike, this is great news!👍
Looking forward to it.
Best
Stephane

silvtal, 2022-09-06:
I want to add that the second approach listed [here](https://support.opentrons.com/en/articles/2519454-using-csv-input-data-in-python-protocols) no longer works.

When you upload a protocol to the latest app version, it does a preliminary analysis from the user's computer, so it needs a file path relative to that computer. However, when trying to run the protocol after that pre-analysis step, it fails because now the robot it's checking its own filesystem. So, even if you use scp and include the file path in the protocol, it won't work because the previous analysis step needs a different path.

This has broken a couple of our pipelines.

I see the only alternative to copypasting the csv into the protocol now is this "experimental HTTP protocol upload", but honestly it looks way more time-consuming than just running a scp command as we usually did.

splaisan, 2024-06-06:
nice efforts but I can only repeat that it would be much more logical to have a data upload/inject button on the robot interface at runtime allowing to replace a placeholder in a template protocol by the content of a csv file hadoc- prepared to match the protocol requirements.
I do this now with an external script hosted on our web-server where we upload both the template protocol without data and the data file and which creates a new ready to go protocol+data that we download then upload on the robot and run (lots of wasted time in the process)
I recently read something in this line about the Flex but will the functionality be accessible for us OT-2 users as well??
Correct me if I am wrong, I did not test the new robot/host-PC software recently.