Opentrons/opentrons #8357: question: can we get the magnetic module disengage lower than default

splaisan, 2021-09-20:
Dear support,

When we disengage, the module descends to a low position then moves back up a few millimeters.
We would like to stay at the lowest position, is this possible and what should be passed using API code to reach this goal?

Thanks in advance

mcous, 2021-09-20:
Please see the documentation for `MagneticModuleContext.engage`: https://docs.opentrons.com/v2/new_protocol_api.html#opentrons.protocol_api.contexts.MagneticModuleContext.engage

closed 2021-09-20 by mcous

splaisan, 2021-09-20:
thanks Mike, 
AFAIK, the page you linked refers to **engage** but it is during **disengage** that I want to remain lower than the default height. It seems that disengage goes down far then comes back up a little; We would like to go down deep and remain there during disengage.

Did I miss something in that page?

Thanks

mcous, 2021-09-20:
Oops, my bad! Read that request too quickly.

The "go down, then back up a little" behavior is observing is the magnetic module hitting its bottom movement limit switch and then backing off the switch as part of its homing procedure. That means the disengage position is the module's lowest reachable point