Opentrons/opentrons #8358: question: custom extension to the python API

splaisan, 2021-09-20:
We have custom functions that we use in almost each python, protocol, they can be considered as custom language extension of the API.
Is there a clean way (preferred location?) to store these functions in a python library file that would be saved on the OT2 PC and sourced in our code to avoid having to copy paste the functions over and over?

Thanks for your help

mcous, 2021-09-20:
Great question! This is an area where we don't have anything really well baked, but we'd like to. So, in that vein: how would you like to deal with - or imagine yourself dealing with - a shared Python library for your organization?

As for existing  "clean" or "preferred" methods to do this, I think a goal with anything Opentrons would build or recommend in this area would need to fit into existing ways to do this in Python. So, I would say a great place to start would be: **use `pip` to manage this shared library.**

If your OT-2 has internet access, this workflow might looks something like this:

1. Structure your shared functionality into a [Python package](https://packaging.python.org/tutorials/packaging-projects/)
2. Publish that package to some sort of artifactory
    - If the code can/should be public, a GitHub repository and/or a [PyPI package](https://pypi.org/) are great options
    - If the code must be private, you should be able to make this work with a private GitHub repository
3. [SSH into your OT-2](https://support.opentrons.com/en/articles/3203681-setting-up-ssh-access-to-your-ot-2) and [`pip install <my_custom_package>`](https://pip.pypa.io/en/stable/user_guide/#installing-packages)

If your OT-2 does **not** have internet access, this workflow can still be applied, with one extra step

1. Structure your shared functionality into a Python package
2. Using your build tool of choice (we're big fans of [Poetry](https://python-poetry.org/) at Opentrons), build a `.whl` file for your pacakge
3. [SCP the wheel file to your OT-2](https://support.opentrons.com/en/articles/3690659-copying-files-to-and-from-your-ot-2-with-scp)
4. Then SSH in and `pip install path/to/my_custom_package.whl`

In both of these workflows, you're going to have the least amount of pain if your custom package **only relies on packages that are already present on the robot**. If your OT-2 has internet access, you can also have declare third-party dependencies, and `pip` will install them for you, as long as those dependencies are pure Python and do not require native compilation.

To see an exhaustive list of all python packages present on your robot, SSH into your OT-2 and run `pip freeze`.

Once your package is `pip install`'d to the OT-2, you can import it into your protocol file:

```python
from opentrons.protocol_api import ProtocolContext
from my_custom_package import my_custom_aspirate

def run(protocol_api: ProtocolContext) -> None:
    # ...
    my_custom_aspirate(pipette=p300_right, volume=50.0)
    # ...
```

splaisan, 2021-09-20:
Thanks Mike,

The option of a pip package is very attractive, only that I never built one and this sounds a bit frightening (bad memories from trying to build a R package years ago).
I will search for a tutorial and see if I could do this, it would indeed be a cool way to also share the work with others (github...)
I was planning to use it exactly as you exemplified in your answer to add customized API commands and other cool little utilities (for instance better formatted log messages, ...)

I was more looking for a way to load/source a local myfunctions.py file put somewhere on the robot filesystem via ssh but I do not want to mess with it and of course may need to redo it after robot upgrades which is not ideal so the pip option seems much cleaner.

I give it a try and see where I end up.
I keep you updated here if I progress....

closed 2021-09-20 by splaisan