Opentrons/opentrons #8385: App: Look through calibration modals and replace the word "even" with "level"

anthonywilleee, 2021-09-22:


SyntaxColoring, 2021-09-29:
This sort of contradicts our support documentation style guide:

> ***flush***: Use *level* instead, like "make sure the tip is level with the top of the well" instead of "make sure the tip is flush with the top of the well." This is based on a user interview with someone who spoke English as a second language. *Flush* meant something different. 💩

The user interview referred to was for the v4.0.0 robot calibration overhaul.

If more recent interviews have shown that *even* is in turn better than *level,* we should update style guidance.

anthonywilleee, 2021-09-29:
I think we're saying the same thing! dont use even, use level. the word "even" wrongly showed up in the user testing mockups. we want to make sure we say level.

SyntaxColoring, 2021-09-29:
Ah wow I must have totally misread this, sorry!

closed 2024-08-13 by sanni-t