Opentrons/opentrons #8388: bug: App and UI: Graphics for labware calibration is inconsistent

alexjoel42, 2021-09-22:
# Overview

When calibrating labware on 96 well plate, tip rack, and reservoir we see an accurate image, but on the tube racks we see the same image as a 96 well plate. This happens for both calibrate to bottom and calibrate to top settings on calibration.

My apologies in advance. I didn't have time to go through all of the labware in our labware library. 

It seems like when the app says "Move to A1":
1. Reservoirs are all represented by nest_12_reservoir_15ml 's graphic (agilent_1_reservoir_290ml and axygen_1_reservoir_90ml are as well) 
2. Tube racks are represented by the 96 well generic image
3. Tip racks and 96 well plates are represented accurately by the 96 well generic image
4. 384 well plates are represented accurately 
5. corning_48_wellplate_1.6ml_flat went to the generic 96 well image 

Image in question:

[image]

# Steps to reproduce
1. Load protocol onto OT-2
2. See correct image
3. Calibrate Labware
4. See incorrect image when calibrating tube racks 

# Current behavior
1. Load protocol
2. Accurate representation on the deck before calibrating
3. Tube rack is shown as a 96 well plate
```Python3
def run(protocol: protocol_api.ProtocolContext):
    tr2 = protocol.load_labware('opentrons_96_tiprack_1000ul', '1')
    p1000 = protocol.load_instrument('p1000_single_gen2', 'right', tip_racks=[tr2])
    reservoir = protocol.load_labware('opentrons_24_aluminumblock_nest_1.5ml_screwcap', '2')
    p1000.pick_up_tip()
    p1000.aspirate(600, reservoir['A2'].bottom(z=2))
    p1000.dispense(600, reservoir.wells()[3])
    p1000.drop_tip()
```

## Example of the deck layout being accurate
[image]

## Example of the labware calibration being 96 wells 

[image]

# Expected behavior
1. Load  protocol 
2. Accurate representation on the deck before calibrating
2. Have the labware be the same image as in the labware library. 
E.G:
```Python3
def run(protocol: protocol_api.ProtocolContext):
    tr2 = protocol.load_labware('opentrons_96_tiprack_1000ul', '1')
    p1000 = protocol.load_instrument('p1000_single_gen2', 'right', tip_racks=[tr2])
    reservoir = protocol.load_labware('nest_12_reservoir_15ml', '2')
    p1000.pick_up_tip()
    p1000.aspirate(600, reservoir['A2'].bottom(z=2))
    p1000.dispense(600, reservoir.wells()[3])
    p1000.drop_tip()
```

[image]

## Risk Assessment 
I've seen one instance of this cause some level of confusion in calibration. Generally though this doesn't come up very often and since the bug applies to many of our labware definitions it could be worthwhile to leave it since this has only come up once in the past year to my knowledge and only tangentially

mcous, 2021-09-22:
Duplicate of #6178

b-cooper, 2021-09-24:
I've updated the name of this ticket to match its duplicate and categorize it as a feature request. This is unfortunately exactly the way that Labware Calibration flow was designed and implemented based. 

 This will be addressed in our current work revamping the protocol upload flow, which is targeting a 4.7 release. This release will include the Labware Position Check which will guide the user through checking the positional accuracy of the robot at each labware included in the protocol.  The analog to this screen will include a dynamically rendered labware as opposed to the few stock images in the current Labware Calibration screen.