Opentrons/opentrons #8423: getting current protocol run:name from inside a function

splaisan, 2021-09-29:
Hi,

Given a minimal script below, how can I get the name **my_protocol** from inside the dummy function (placeholder <######>) without having to name it explicitly?

Such facility would allow reusing universal custom functions based on API functions without risking to wrongly name them (eg. **protocol** or **ctx** depending on the recycling source).

This is possibly a dummy question, please be kind!

My current solution is to reference **my_protocol** and use **funcctx** in the function in place of  <######>

```
funcctx = my_protocol
```

But I dream of something even more straightforward :-) with some `self.` or `_.` magic-bullet

Thanks in advance

```
from opentrons import protocol_api

metadata = {'apiLevel': '2.10'}

# dummy code to support my request

def run(my_protocol: protocol_api.ProtocolContext):
    # Labware defining tip racks
    tr1 = my_protocol.load_labware(
        'opentrons_96_tiprack_20ul',
        '10')

    # Defining Pipette
    p20_single = my_protocol.load_instrument(
        'p20_single_gen2',
        'right',
        tip_racks=[tr1])

    # Block Command picking up tip
    p20_single.pick_up_tip()

    def my_dummy(text):
        <######>.comment(text)

    my_dummy('Hello world!')
```

SyntaxColoring, 2021-09-29:
I would write it like this:

```python
def run(my_protocol: protocol_api.ProtocolContext):
    # Labware defining tip racks
    tr1 = my_protocol.load_labware(
        'opentrons_96_tiprack_20ul',
        '10')

    # Defining Pipette
    p20_single = my_protocol.load_instrument(
        'p20_single_gen2',
        'right',
        tip_racks=[tr1])

    # Block Command picking up tip
    p20_single.pick_up_tip()

    my_dummy(my_protocol, 'Hello world!')

def my_dummy(protocol: protocol_api.ProtocolContext, text: str):
        protocol.comment(text)
```

In other words, it's the caller's responsibility to supply the correct thing as the `ProtocolContext`. The custom function (`my_dummy()`) doesn't attempt to figure it out on its own.

splaisan, 2021-09-29:
Thanks a lot Max for your example and clarification,

As said, I am new to python and the code layout is still not entirely clear to me.

I had seen the `def get_values(*names): ` block in protocols but did not realize that you could also have def blocks after of the run def block, thought there was some kind of linear precedence as in bash functions.

From your edited example, I understand that all def blocks (before or after the run block) will be _evaluated_ before the main run block is executed (am I right?).

<maybe one day I will be fluent in python ...>
:-)

closed 2021-09-29 by splaisan

SyntaxColoring, 2021-09-29:
> From your edited example, I understand that all def blocks (before or after the run block) will be _evaluated_ before the main run block is executed (am I right?).

Yeah, both of these are effectively equivalent in Python:

```python
def foo():
    return 123

def run(protocol: protocol_api.ProtocolContext):
    x = foo()
```

```python
def run(protocol: protocol_api.ProtocolContext):
    x = foo()

def foo():
    return 123
```

It works something like this:

1. We load your protocol file as a Python module.
2. The Python interpreter goes through the file and makes sure there are no syntax errors.
    * Note that in Python, referring to a name that does not exist yet is usually not a syntax error. It's a runtime error.
3. The Python interpreter executes all the module's top-level statements, top to bottom. The function definitions `def foo(): ...` and `def run(): ...` are top-level statements. When the Python interpreter "executes" those function definitions, it only makes the names `foo` and `run` accessible to subsequently-run code. It does *not* run the actual code inside those functions yet.
4. We call your `run()` function.
5. Your `run()` function goes to call `foo()`, so it looks up the name `foo` in the current module, which works because that was set up in step (3).

If it's clearer to you, you can always keep `foo()` above `run()`, of course.

Let me know if this helps!

splaisan, 2021-09-29:
Very informative!
I had no idea that it was working that way.
thanks a lot for your info