Opentrons/opentrons #8432: bug:

SyntaxColoring, 2021-09-29:
# Overview

The Opentrons App has a **Browse** button to manually send an `ot2-system.zip` system update file to the OT-2. For robustness, this is supposed to work even when the robot server on the robot has crashed. Only the update server needs to be working.

However, even when the update server is working and you *should* be able to do this, the Opentrons App sometimes grays out the **Browse** button and makes it unusable.

# Steps to reproduce

1. Connect to an OT-2 with the Opentrons App. You don't actually need to click the blue toggle button; the robot just needs to be showing up in the left sidebar with the toggle button *available* and without any orange error banner.
2. Observe that the manual update **Browse** button is clickable.
3. `ssh` into the robot and run `systemctl stop opentrons-robot-server`.
4. Wait for an orange error banner to appear in the Opentrons App.
5. Observe that the **Browse** button is now grayed out and unclickable.
6. Close and reopen the Opentrons App.
7. The orange error banner will still be there. However, observe that the **Browse** button is available again.

# Current behavior

It seems like the Opentrons App will incorrectly gray out the **Browse** button when the robot server goes down *when it was not down before.* If the robot server is down when the Opentrons App first discovers the robot, it seems like things work correctly.

# Expected behavior

As long as it looks like the update server is able to receive an `ot2-system.zip` file, the **Browse** button should be usable.

SyntaxColoring, 2021-09-29:
Running through the above steps to reproduce again, I couldn't get the **Browse** button to gray out. There might be something else going on here.

I am positive that I've had the **Browse** button be grayed out and have "fixed it" by closing and reopening the Opentrons App, which shouldn't work.

mcous, 2021-10-04:
This button's disable logic was previously broken (#5429) and fixed (#6483). The fix contained most of the "is this button disabled?" logic to a single function:

https://github.com/Opentrons/opentrons/blob/63eb7641b2690fdd559dd38b8d0ecd8242d30bd5/app/src/redux/buildroot/selectors.ts#L136-L173

This logic looks pretty robust to me, so I'm sort of scratching my head as to why you would see this behavior. Did you check what the tooltip said?

We've had stale data problems of this type in the discovery client before, so I wouldn't rule out the DC accidentally marking the update server's health as bad given some set of conditions

SyntaxColoring, 2021-10-04:
> Did you check what the tooltip said?

I did, but I didn't write it down, and I don't remember. :(