Opentrons/opentrons #8474: HTTP API: persist previous sessions

mcous, 2021-10-06:
## Overview

As an HTTP API client, I'd like to retrieve a history of previous run sessions, so that I can inspect and re-create a given run.

In order to satisfy the needs of the Protocol Upload Revamp project, the new ProtocolEngine-based `/sessions` endpoints need the following changes:

1. More than one session should be allowed
2. A session can be created from the state of another session
3. Previous sessions' state should be persisted through reboot

This ticket covers item (3). See #8470 for item (1) and #8473 for item (2).

## Acceptance Criteria

- [ ] When a protocol session becomes stopped, its state is persisted to disk
- [ ] When a protocol session becomes stopped, its ProtocolEngine state is also persisted to disk

Writing to disk every time a session changes would be... a lot, so I think for our MVP we can leave this as "when the protocol ends or is stopped," and leave "unexpected shutdown during a run" undefined.

## Implementation Notes

TBD

TamarZanzouri, 2022-04-05:
@mcous RunResource contains a prop of actions. it is a list on RunActions which contains id, createAt and RunActionType (str\enum). we have a few ways to conquer this. we can store it all in a RunActions table with a reference to the run or we can store it as a blob object (in case we can assume it wont change a lot). would love to hear your take about this

TamarZanzouri, 2022-04-06:
create_run_action -> to update a run

TamarZanzouri, 2022-04-06:
create_run_action, update_run -> update a current run

TamarZanzouri, 2022-04-13:
@mcous looks like an acceptance criteria for this task is to update the run notes.
Should I add this? or should we leave it for 5.2?

TamarZanzouri, 2022-04-19:
@mcous Meeting notes 4/19/2022

TamarZanzouri, 2022-04-19:
**### Meeting notes 4/19/22:**
@mcous Following our meeting. Please let me know if there is anything I am missing.

- Persist base_router.py/update_run ->  engine_store.get_state(run.run_id) and actions_router.py -> protocol_runner.run 
- Use ProtocolRunData, move to shared location. add labware_offsets from engine_state.labware.get_labware_offsets() 
- Store state as one blob using pydantic models
- Create a CommandStore inside robot-server commands directory 
- Make sure when translating to json that the order is preserved - if its lists it will be - if its dicts it prob won’t
- EngineStore stores a dictionary[run_id, protocol_engine]. We might be able to removed it after persisting to the DB. 
- ProtocolEngine commands in the CommandsState are associated to the run in the engine_store

PR #10187 (2022-05-17): feat(robot-server): persist previous run results and commands
# Overview

Persisting protocol engine state to the DB on run pause and on run update. closes #8474

# Changelog

- moved ProtocolRunData to protocol_engine/state
- added method get_protocol_run_data to StateView
- insert ProtocolRunData and Commands into DB on route `/runs/{runId}` update to not current
- insert ProtocolRunData and Commands into DB on route `/runs/{runId}/actions` when calling `engine_store.runner.run()`
- get all commands and get commands slice on route '/runs/{runId}/commands'
- get command by id on route '/run/{ [... cut]
commit 87677c69 (2022-05-17): feat(robot-server): persist previous run results and commands (#10187)

Closes #8474
closed 2022-05-17 by mcous