Opentrons/opentrons #8489: bug: Pausing directly after a pause will skip all but the last pause commands.

Zanecrc1, 2021-10-07:
# Current behavior

Using the pause() command directly after a pause() will skip all but the last pause. 
```
    ctx.pause('Pause 1')
    ctx.pause('Pause 2')
    ctx.pause('Pause 3')
    ctx.pause('Pause 4')
```
Will only output 'Pause 4'

# Expected behavior

Each pause command should be executed in order

mcous, 2021-10-08:
@andrewchastain not easily in the current system, no. There's a chain of private property accesses you could use, but that is not at all stable. The best available workaround is to insert a dummy command inside the pauses:

```python
# haven't tested, but I _think_ this would work
ctx.pause("Pause 1")
ctx.set_rail_lights(on=False)
```

The root of the issue is that in PAPIv2, pauses are fundamentally controlled by the hardware control layer, and there is effectively no protocol execution layer. This makes this behavior too fraught to adjust in PAPIv2. Marking this as `papi-v2-wontfix`, accordingly

andrewchastain, 2021-10-08:
Yes, if I put `.move_to()` in between that makes both pause steps work, even if there isn't movement. Is there a way to access the current location of the gantry (like `types.Location` or `.location_cache`)?

andrewchastain, 2021-10-08:
Yes, there have been a few situations where `.location_cache` gave an unexpected result. If the last step before the call was to `return_tip()` it causes the pipette to move down as far as possible.

andrewchastain, 2021-10-12:
Thanks. I noticed that after a call of `.return_tip()` the `location_cache` has a z value of 24.43, so I have to make sure that each `.move_to()` calls something like the `move_target` that you mentioned before `location_cache` if the last move was returning the tip.