Opentrons/opentrons #8496: Spike: evaluate robot-server data persistence via sqlite

mcous, 2021-10-08:
## Overview

The OT-2's HTTP API needs to persist various resources across reboots, for example, uploaded protocols and previous runs (#8495). In the past, this sort of persistence has been accomplished two ways:

- A sqlite database
    - Protocol API v1 labware definitions and calibrated positions
- JSON on disk
    - Pretty much everything else

For persistence of newer HTTP resources, we need to decide which of these two paths we'll take

## History 

The database table for v1 labware definitions had serious schema problems, which caused it to be very difficult to work with. For example, "calibrating" a labware would permanently mutate a labware record, such that its original, non-calibrated position was lost. Additionally, its access code was poorly implemented, leading to unnecessarily bad performance.

These problems were addressed by v2 labware definitions, which also went ahead and threw away the sqlite database. More or less, this spike exists to evaluate whether sqlite (or an actual relational database of some sort), when properly used, is actually beneficial for our software. Indications that it may be:

- Our data is pretty highly relational
    - e.g. a run resource refers to a protocol file resource which refers to a protocol analysis resource
- Nosql generally in my experience leads to a bunch of one-off / divergent ways of modeling relationships that shift over time
- JSON-on-disk specifically has been implemented a slightly different way every time we have a new resource to persist, leading to confusion and high maintenance costs

## Acceptance Criteria

One of the following:

- A proposal + tickets to add a sqlite database to the robot-server, with
    - Typed data classes (e.g. via Pydantic)
    - Migration capabilities
- A proposal + tickets to pull in [junk-drawer](https://github.com/Opentrons/junk-drawer), so that we have JSON-on-disk persistence with
    - Typed data classes via Pydantic
    - Migration capabillities 

## Data layer needs

- For our purposes, an ACID (Atomic, Coherent, Isolated and Durable) data layer is preferable
    - Data should be readable as soon as it is written
    - Updates to multiple related resources should fail completely if any one update fails; otherwise our data wouldn't make sense
- Raw data should be inspectable by the user and support

mcous, 2022-03-08:
## Spike results

- Upon investigation and consultation with other engineers, **we expect SQLite to be a good solution to our persistence needs**
- I will add the following library recommendations to #8474 and #8468

## Recommendations

- [databases](https://www.encode.io/databases/) - A lightweight library for async SQL DB access, using queries from [SQLAlchemy Core](https://docs.sqlalchemy.org/en/latest/core/)
    - Taking some lessons from our stuggles with with FastAPI and Pydantic, I think we should keep things lightweight for now
    - I evaluated these other libraries/tools, that I thought were all either too heavy or not ready for primetime: [SQLModel](https://sqlmodel.tiangolo.com/), [SQLAlchemy](https://www.sqlalchemy.org/), [Pydantic ORM mode](https://pydantic-docs.helpmanual.io/usage/models/#orm-mode-aka-arbitrary-class-instances)
- [Alembic](https://alembic.sqlalchemy.org/en/latest/) - SQL migration tool

closed 2022-03-08 by mcous

SyntaxColoring, 2022-03-11:
Further investigation:

* I agree we should at least use SQLAlchemy Core.
* It’s unclear to me if we should avoid SQLAlchemy’s ORM. Perhaps we should use SQLAlchemy ORM in addition to SQLAlchemy Core.
    * Briefly playing around with it, with just Core, I don’t see how a static type-checker could catch typing errors. I think that results from a `select` will get typed as `Any`, and input to an `insert` is statically allowed to mismatch the type specified in the table metadata.
    * Note that SQLite itself [also plays fast and loose with types](https://www.sqlite.org/flextypegood.html). So typing bugs won’t even necessarily get caught at runtime unless we’re careful to do something about it.
    * I haven’t tested if using the ORM is actually any better than this, but theoretically it could be, I think. [They have a mypy plugin for it.](https://docs.sqlalchemy.org/en/14/orm/extensions/mypy.html)
* I think we can drop the dependency on [encode/databases](https://www.encode.io/databases/), for simplicity. I don’t see it adding much.
    * [SQLAlchemy already has official asyncio support](https://docs.sqlalchemy.org/en/14/orm/extensions/asyncio.html). It’s beta, but encode/databases is v0.5.5, so we’re a little screwed either way. ¯\_(ツ)_/¯
    * I personally find database’s connection lifetime model a bit confusing. Same with [its strategy for test isolation](https://www.encode.io/databases/tests_and_migrations/). From what I can tell, plain SQLAlchemy already does both of those those things well, so no need to reinvent it.