Opentrons/opentrons #8625: Analysis sometimes hangs after uploading a protocol over HTTP

SyntaxColoring, 2021-10-29:
Pulling this out of #8557, where the problem was originally noticed. I don't *think* that PR introduced this problem, but it's possible.

-----

Testing on a robot revealed a bug somewhere:

* With the "Use Older Protocol Analysis Method" feature flag **enabled** (i.e., *without* "fast simulation"), the server analyzes my test protocol correctly.
* With that feature flag **disabled** (i.e., *with* "fast simulation,") that same protocol's analysis is stuck in the "pending" stage indefinitely.
* Tailing `journalctl` logs suggest that the protocol *is* running to completion: a `protocol.comment("done")` appended to the end of the protocol is reached successfully. So for some reason, despite that, the server isn't recognizing that the protocol is done?
* `systemctl stop opentrons-robot-server` hangs until a timeout while it waits for background tasks to complete, which suggests that the background analysis task is never finishing. See also #8588.

Seems to be hanging here:

https://github.com/Opentrons/opentrons/blob/549887950be5d90d3e5f08a598b7a95cebf45978/api/src/opentrons/protocol_runner/task_queue.py#L79

amitlissack, 2021-11-04:
I managed to make a small code change that makes the bug happen consistently.

I added a sleep in `StateView.wait_for`:
```
        while not is_done:
            await asyncio.sleep(1)
            await self._change_notifier.wait()
            is_done = predicate()
```

I believe this is the culprit in ChangeNotifier.wait
```
 self._event.clear()
 await self._event.wait()
```

We are blindly clearing the event and then waiting for it.

There smells like a hole there. Like it’s possible that the event is already set (for the StopAction) and we’re clearing it.

I believe that inserting the await asyncio.sleep(1) shouldn’t have any impact on the wait algorithm.

mcous, 2021-11-09:
Is there a reliable reproduction protocol that doesn't involve changing the engine code to trigger the bug?

commit d55335c0 (2021-11-11): refactor(engine): add threadsafe wrappers to legacy plugin dispatches (#8701)

Closes #8625
closed 2021-11-11 by mcous