Opentrons/opentrons #8635: bug: UnicodeEncode Error in Jupyter Notebook simulation on a Korean Windows 10 laptop

Matt-Zwimpfer, 2021-11-03:
# Overview
When executing the following type of simulation code in a Jupyter Notebook cell of a Korean Windows 10 laptop: 
`!opentrons_simulate /path/to/protocol -L /path/to/custom_labware`

An error arises: 
`UnicodeEncode Error: cp949 codec can't encode character '₩xb5' in position 630: illegal multibyte sequence" `

This error doesn't pop up when the same code is simulated on a Windows 10 or a Mac. 

# Steps to reproduce
1. Have a Korean Windows 10 laptop with Jupyter Notebook running Python version 3.8.5 installed
2. Try to simulate a Python protocol, which references custom labware, with this type of code: `!opentrons_simulate /path/to/protocol -L /path/to/custom_labware`
3. UnicodeEncode error should pop up

# Current behavior
Throwing UnicodeEncode Error that prevents simulation of the protocol when executed in a Jupyter Notebook running Python 3.8.5 on a Korean Windows 10 laptop

# Expected behavior
Protocol should simulate and output a run log of the protocol steps

Matt-Zwimpfer, 2021-11-03:
### Current Workaround
Should be able to run the same code on the command line to simulate the protocol and have the run log output.

SyntaxColoring, 2023-06-28:
I think this is because the run log contains "µ" characters (as in "aspirating 123 µL"), which I guess aren't supported by the [cp949](https://en.wikipedia.org/wiki/Code_page_949_(IBM)) encoding.

```python
>>> "µ".encode("cp949")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
UnicodeEncodeError: 'cp949' codec can't encode character '\xb5' in position 0: illegal multibyte sequence
```

The error happens when `opentrons_simulate` tries to print that character to its stdout.

Here's what [the Python docs](https://docs.python.org/3.7/library/sys.html#sys.stdout) say about the encoding of stdout:

> On Windows, UTF-8 is used for the console device. Non-character devices such as disk files and pipes use the system locale encoding (i.e. the ANSI codepage).

This could explain why it works when you run `opentrons_simulate` directly in a terminal, but not when you run `!opentrons_simulate` inside a Jupyter Notebook cell.

To fix this, the `!opentrons_simulate` subprocess needs to have its stdout encoding set to UTF-8 (or any encoding supporting the µ character), and Jupyter Notebook needs to match that when it renders the output in a cell. I'm not sure how to configure this in Jupyter Notebook. Wild guess: Maybe if you launch it with `PYTHONIOENCODING=utf-8 jupyter notebook`?

Since this doesn't appear to be a bug with Opentrons software, I'm closing this. But if anyone figures out how to fix the Jupyter encoding problem, please let us know here. I'm curious!

closed 2023-06-28 by SyntaxColoring