Opentrons/opentrons #8651: Remove all spaces between volumes and units

mikecardarelli, 2021-11-05:


mikecardarelli, 2021-11-09:
@Kadee80 After talking with Morgan, I'm not sure we still want to make this change. Before you start on this, let's discuss quickly on Slack/Hangouts. Sorry for the confusion

closed 2021-11-10 by mikecardarelli

mcous, 2021-11-16:
FWIW I support leaving this one alone! A space between numbers and their corresponding SI units is [recommended by NIST](https://physics.nist.gov/cuu/Units/checklist.html), and was why we included the space in the first place (if I'm remembering that hazy time correctly).

If we're concerned about line-wrapping, we can replace the space between the number and the units with a [non-breaking space character](https://en.wikipedia.org/wiki/Non-breaking_space)