Opentrons/opentrons #8658: API v2 documentation navigation problem

danolson1, 2021-11-08:
When reading the [API v2 documentation](https://docs.opentrons.com/v2/) on my phone, it would be convenient to have a link at the bottom of one section to allow you to jump right to the next section without having to navigate back to the table of contents.

For example, at the bottom of the [Pipettes ](https://docs.opentrons.com/v2/new_pipette.html)section, there should be a link to go to the [Building Block Commands](https://docs.opentrons.com/v2/new_atomic_commands.html) section.

SyntaxColoring, 2024-01-02:
@ecormany Any opinions on this?

> For example, at the bottom of the [Pipettes ](https://docs.opentrons.com/v2/new_pipette.html)section, there should be a link to go to the [Building Block Commands](https://docs.opentrons.com/v2/new_atomic_commands.html) section.

I'm not sure if/when this changed, but it looks like if you view docs.opentrons.com on a phone now, the *full* table of contents is at the bottom of the page. So in a sense, the link *is* there already (😛) but dedicated previous/next links might still be helpful. See e.g. https://react.dev/learn/writing-markup-with-jsx for an example of this in the wild.

ecormany, 2024-01-02:
I don't feel strongly about including or omitting these links, as long as they're automatic. There is no promise that the linear order of the docs won't change over time (and, in fact, they are becoming less linear as large topics get broken into subpages).

I think that the change in mobile display happened with #13349. That level of hacking is what I know can be done with our current Sphinx template, Alabaster. However, I think we may be outgrowing its abilities and I'd like to explore some broader design improvements. 

I've noted this feedback internally as RTC-380, part of a larger loose collection of similar potential improvements (RTC-375).

ecormany, 2024-01-02:
Further detail: this is [included in Alabaster](https://github.com/sphinx-doc/alabaster/pull/101), but I still want to address it at the same time as other design changes.