Opentrons/opentrons #8782: feat: Adding more flexibility for pauses and delays

alexjoel42, 2021-11-16:
# Overview

The problem statement is currently how to best integrate manual steps into an Opentrons workflow. Currently with a delay/pause folks need to use a stop watch to track timing for when to do manual steps. 

[image]

Use this section to describe your issue at a high level. Please include some context about why you want this feature. What would this feature allow you to accomplish? Are there any workarounds you use to partially achieve your goal? Are there any issues you could find that may be related?
-->

# Implementation details

I would defer to the user who recommended this feature, but I'd say this could be addressed in two ways:

1.  Having a clock under the runtime tracker for longer pauses/delays (+10 minutes).
2. What we could do is add an alert function potentially in the middle of a [delay](https://github.com/Opentrons/opentrons/blob/edge/api/src/opentrons/protocol_api/protocol_context.py#L733-L742)

# Design

Feasibility it would possibly look something like 

```Python3 
import subprocess
from opentrons import protocol_api, types

metadata = {'apiLevel': '2.8'}

metadata={"apiLevel": "2.9"}

AUDIO_FILE_PATH = '/etc/audio/speaker-test.mp3'

def run_quiet_process(command):
     subprocess.check_output('{} &> /dev/null'.format(command), shell=True)

def test_speaker():
     print('Speaker')
     print('Next\t--> CTRL-C')
     try:
         run_quiet_process('mpg123 {}'.format(AUDIO_FILE_PATH))
     except KeyboardInterrupt:
         pass
         print()

def alarm_delay(self, seconds=0, minutes=0, msg=None, alert = True, alert_time = seconds):
       """Delay protocol execution for a specific amount of time.
       :param Boolean alert: Whether someone wants an alert or not 

       If `seconds` and `minutes` are specified, they will be added.
       if 'Alert' is not specified default to false
       if 'alert_time' is not specified default to 0
       """
       self._implementation.delay(seconds=delay_time-alert_time, msg=msg)
      if alert_time > 0:
         self._implementation.delay(seconds = alert_time, msg = f"the protocol is {alert_time} away from completion")
         if not protocol.is_simulating:
            test_speaker()
        else: 
          pass 

```

# Acceptance criteria

I would have to research more on the use cases whether it is more of an API feature request to address the issue. The question would mostly come down to efficiency since this might be easier to implement than an app feature. 

 (

danolson1, 2022-04-14:
I have a similar issue. I have a 2+ hour protocol with some critical timing steps at the very end that require manual intervention. It would be great if the pause() function gave a more obvious visual indication that the protocol is paused. For example, including an option to blink the deck lights until the protocol is resumed.

-Dan

wolfrickw, 2025-04-04:
I had a similar issue, i really needed some alarm/visual indication of pause. Obviuously no useful help from opentrons (i was waiting for month...). I tried every firmware (now i went back to 4.7) without success, so I just mounted a buzzer and a switch in paralell with the led lights and it works well. What I just learned during my journey, that I wouldn't ever again upgrade to any firmware... as the version number incerease a number of useless functions are implemented and some basic functions are broken....
W

alexjoel42, 2025-04-04:
Have you tried following [this article for the OT-2 ](https://support.opentrons.com/s/article/How-to-play-sounds-out-of-the-OT-2) ? 

I hope the article helps you along the way. 

I think what I wrote way back when should work on 4.7.0 if that article was updated for newer software versions 🤞 .

I'm sorry that you've had a frustrating experience.