Opentrons/opentrons #8973: bug: Pipette offset can't be calibrated on 5.0.0-beta.0

SyntaxColoring, 2021-12-02:
# Steps to reproduce

1. Update your Opentrons App and robot to v5.0.0-beta.0.
2. Attempt tip length calibration, either through **Robot > Pipettes** or through the **Protocol** tab.

# Current behavior

After completing the calibration flow, the pipette still shows up in the app as needing calibration.

There is no change if you close and reopen the app.

This might happen for tip length calibration in addition to pipette offset calibration. I haven't been able to get that far to test.

# Expected behavior

After completing the calibration flow, the pipette should show up as calibrated.

SyntaxColoring, 2021-12-02:
The error was resolved for me by doing a factory reset through the app, selecting tip length calibrations, pipette offset calibrations, and labware calibration.

Investigating now:

* Whether this was a problem before the beta
* What, exactly, was wrong on the filesystem

Laura-Danielle, 2021-12-06:
@SyntaxColoring, do you recall which robot you were able to reproduce this on? Typically the error will "persist" if there was an unsupported file to begin with from a previous SW version that caused the issue (i.e. a `none.json` file). We could be more strict in what we will accept as a valid pipette for in the calibration flows -- right now, we're only checking that a pipette object exists on a given mount.

SyntaxColoring, 2021-12-06:
@Laura-Danielle Yeah, it's the robot in the Neutrino room. On that robot, I saved some of the apparently-corrupted data before factory resetting. I think it's under `/data/robot-old-<date>` or something. `None.json` looks familiar; I bet that's it.

If this does turn out to be the problem (which it really looks like it is), I'm fine with the solution just being to clear out the calibration files.

SyntaxColoring, 2021-12-06:
I've confirmed that this robot had `null`/`None` references in `/data/robot/pipettes` and `/data/tip_lengths`. I'm closing this issue, assuming it's a manifestation of the already-fixed #8576. Thanks for your help @Laura-Danielle!

Just in case, here are archives of `/data/robot` and `/data/tip_lengths` as they were found on the robot at the time of the problem:

* [robot.old-2021-12-02.zip](https://github.com/Opentrons/opentrons/files/7662587/robot.old-2021-12-02.zip)
* [tip_lengths.old-2021-12-02.zip](https://github.com/Opentrons/opentrons/files/7662586/tip_lengths.old-2021-12-02.zip)

closed 2021-12-06 by SyntaxColoring