Opentrons/opentrons #8975: Rename or remove labware calibration from factory reset dialog

SyntaxColoring, 2021-12-02:
In the Opentrons App, the "Factory Reset" dialog currently has a "Labware Calibration" option. This won't make sense past v5.0.0, which replaces labware calibration with the Labware Position Check.

* Decide whether we want to retain this factory reset checkbox despite labware calibration not existing anymore. If we're keeping the files around on the filesystem past v5.0.0 for downgrade purposes, it might also make sense to keep the checkbox around.
* Either remove that checkbox, or, if we're retaining it, consider renaming it to clarify that the offsets are legacy-only.

emilywools, 2021-12-06:
hey @SyntaxColoring we are going to remove this checkbox per #8283

SyntaxColoring, 2021-12-06:
👍 Closing as a duplicate of #8283.

closed 2021-12-06 by SyntaxColoring