Opentrons/opentrons #8977: Stub for Max to fill out about module compatibility

b-cooper, 2021-12-03:


SyntaxColoring, 2021-12-04:
Rough solution directions off the top of my head. This list isn't exhaustive.

* We change the Protocol Engine `loadModule` command contract, so that instead of loading an exact, strictly matching `moduleModel`, it's allowed to load any "compatible" model.

   What counts as "compatible" is left for the robot software to decide. 

   The command result would probably include the actually-loaded model, while the requested model could still be retrieved from the request params.

   Internally in the robot software, we update the PAPIv2 legacy command mapper to correctly pass along the requested model *and* loaded model.
   
   We change the labware offset contract so that the client specifies the model that it expects will be *requested,* instead of the model that it expects will be *loaded.*

   We update step (4) so that the server compares the requested model, not the loaded model.

* Something involving an explicit server-provided link between a labware offset and an analyzed resource that the offset is intended to apply to. For example, stable IDs, but there are also other ways of doing this.

mattwelch, 2021-12-28:
Closed in favor of #8914

closed 2021-12-28 by mattwelch