Opentrons/opentrons #8985: Performance problems when multiple clients are connected to the HTTP API

SyntaxColoring, 2021-12-06:


SyntaxColoring, 2021-12-06:
Avenues for improvement:

* Use `pydantic`'s `.construct()` where appropriate to avoid unnecessary validation
* Consider doing blocking copies (?) in threads instead of the main event loop
    * We'll have to be careful with this to do it in a thread-safe way

shlokamin, 2021-12-06:
[post from conversation with @SyntaxColoring about an issue that seems related]

When uploading the attached python protocol with a TC, analysis starts, and then shortly after the robot disconnects from the client (we see the disconnect modal in the app). When reconnecting to the robot later, I can see that analysis is complete, and all of the accordion steps are correctly rendered. 

When I start LPC, the network requests to execute the setup + initial movement commands hang, the robot disconnects from the client, and then shortly after the server executes the setup + initial movement commands

[pcr_optimization.py.zip](https://github.com/Opentrons/opentrons/files/7663354/pcr_optimization.py.zip)

mattwelch, 2021-12-28:
Closing in favor of other tickets which capture this work.

closed 2021-12-28 by mattwelch