Opentrons/opentrons #8986: feat(engine): add `key` hash to command for analysis-to-run matching

mcous, 2021-12-06:
## Overview

As an HTTP API user, I'd like to associate simulated commands from a `/protocols` analysis with the actually run commands in a `/runs` run.

## Acceptance criteria

We'd like to add a `key` (or other appropriate name, e.g. `hash`) field to a `Command` resource. This matching key has two requirements:

- A given command's match key is unique in the protocol
- In a deterministic protocol, a command's match key is stable across an analysis and a run 

## Implementation details

A hashing of the following items should do the trick

- Command type
- Command order in the stream
- Command params

### Hashing the command order

TODO

### Hashing the command params

TODO

### Why not use the command ID?

A command's ID represents something subtly different: it's a unique identifier for a single instance of a command. A command that was executed in an analysis is a different instance than one that was executed in a real run; if nothing else, the execution environment is vastly different.

Additionally, an ID is (read: will be) used for record persistence. Persisting things becomes much less technically fraught if we know that child command resources owned by different parents (like an analysis resource or a run resource) won't have ID collisions in some database table.

SyntaxColoring, 2022-01-20:
We might also want to include the previous command hash as an input to the current command hash.

So if analysis does:

1. Pick up tip
2. Aspirate from A1
3. Dispense in A2
4. Aspirate from B1

And the actual run does:

1. Pick up tip
2. Aspirate from *C1*
3. Dispense in *C2*
4. Aspirate from B1

The run key for step (4) will *not* match the analysis key for step (4), reflecting that the physical and logical state of the robot has permanently diverged.

mcous, 2022-06-17:
Ticket under investigation as a potential 6.1.0 release requirement

mcous, 2022-06-23:
Not a 6.1.0 release requirement. Command key additions required for the 6.1.0 release are tracked in #10799

mcous, 2022-11-02:
Moved to RCORE-326

closed 2022-11-02 by mcous