Opentrons/opentrons #8996: bug: Can't close protocol after protocol fails

SyntaxColoring, 2021-12-06:
As reported by @reachojhaneha for v5.0.0-beta.0:

> unable to close the protocol once the protocol failed. Had to close the app, screen froze. I had to restart the robot, restarting app also did not help.

shlokamin, 2021-12-07:
I'm seeing this intermittently as well. I can't seem to reliably reproduce, but it happens when I'm in the following state:
1. I've started a run, and cancel it after a few movement commands 
2. the run is still `current`
3. the run is in a `stopped` state

When i try to close the run (using the X button in the protocol tab), i get a two `409` HTTP conflict errors:
1. when trying to uncurrent the run by issuing a `PATCH` to the run's endpoint
2. when trying to delete the run by issuing a `DELETE` to the run's endpoint 

It's happened a few times using the `mixSettings` protocol in `shared-data`

mikecardarelli, 2021-12-14:
@reachojhaneha We believe some recent CPX would should have fixed this. Could you try to repro this on the latest build when you have a chance? Thanks!

mattwelch, 2021-12-28:
Closing as resolved

closed 2021-12-28 by mattwelch