Opentrons/opentrons #9013: bug: beta0- App goes blank when  Run Again button is clicked

reachojhaneha, 2021-12-07:
Steps to reproduce:
1. Run a protocol
2. Once the protocol is complete -> "Run Again" button appears
3. Upon clicking the button "Run Again" -< app goes blank

shlokamin, 2021-12-07:
hey @reachojhaneha! can you please add the protocol you used here?

shlokamin, 2021-12-08:
More info from @sanni-t:

> I was just testing the cancelled error PR using the app and it looks like after a successful cancellation, if I try to run again then the app becomes blank. I see this weird request in the network logs: http://192.168.0.24:31950/runs/undefined 

protocol sanniti was running:
[simple_transfer_proto.py.zip](https://github.com/Opentrons/opentrons/files/7677379/simple_transfer_proto.py.zip)

Note that the first command is a protocol pause, which the app doesn’t know of due to a known bug, so you’ll have to click the pause button and then resume.

reachojhaneha, 2021-12-09:
@shlokamin , i am able to continously able to reproduce this issue, let me know if you would like to pair and get some logs

shlokamin, 2021-12-09:
@reachojhaneha I'm unable to reproduce locally, can you please upload the protocol you used?

shlokamin, 2021-12-10:
This is not actually an issue, dependencies just needed to be updated. Closing

closed 2021-12-10 by shlokamin