Opentrons/opentrons #9059: add a way for SSH/Jupyter users to download offset data

emilywools, 2021-12-09:


mcous, 2022-01-12:
Proposal for the copy-paste block!

1. Enumerate all labware checked by LPC
2. Follow each load with a call to `labware.set_offset`

For example, given the following LPC:

| labware/module                           | location         | calculated offset |
| ---------------------------------------- | ---------------- | ----------------- |
| `temperatureModuleV1`                    | Slot 3           | N/A               |
| `opentrons_96_tiprack_300ul`             | Slot 1           | x: 1, y: 2, z: 3  |
| `nest_96_wellplate_100ul_pcr_full_skirt` | Module in slot 3 | x: 4, y: 5, z: 6  |

### Jupyter

```py
import opentrons.execute
protocol = opentrons.execute.get_protocol_api("2.12")

module_1 = protocol.load_module("temperatureModuleV1", location="3")

labware_1 = protocol.load_labware("opentrons_96_tiprack_300ul", location="1")
labware_1.set_offset(x=1.0, y=2.0, z=3.0)

labware_2 = module_1.load_labware("nest_96_wellplate_100ul_pcr_full_skirt")
labware_2.set_offset(x=4.0, y=5.0, z=6.0)
```

### `opentrons_execute` CLI (aka SSH)

```py
from opentrons import protocol_api

metadata = {
    "apiLevel": "2.12"
}

def run(protocol: protocol_api.ProtocolContext):
    module_1 = protocol.load_module("temperatureModuleV1", location="3")

    labware_1 = protocol.load_labware("opentrons_96_tiprack_300ul", location="1")
    labware_1.set_offset(x=1.0, y=2.0, z=3.0)

    labware_2 = module_1.load_labware("nest_96_wellplate_100ul_pcr_full_skirt")
    labware_2.set_offset(x=4.0, y=5.0, z=6.0)
```

Not sure about the whole `labware_1`, `module_1` naming scheme, but the important points here are:

- Must use API version `2.12` or higher
- Make sure to include both deck labware and module labware

b-cooper, 2022-01-18:
This task should be broken into three separate PRs:

- create an advanced setting for the app for this to hide behind
- create a link with modal that only shows the jupyter snippet in a textarea
- add the ssh/cli snippet to that modal

commit 6ae7f925 (2022-01-25): refactor(app): add advanced setting gated link to raw labware offset data for jupyter/cli execution

Include generated labware load and offset snippets of python that can be copied and pasted into
jupyter or cli protocol execution. Link is only accessible behind an app advanced setting.

Closes #9059
PR #9311 (2022-01-27): refactor(app): add advanced setting gated link to raw labware offset data for jupyter/cli execution
# Overview

Include generated python snippets that load labware/module and apply offsets in the labware setup step. Link is only accessible behind an app advanced setting.

Closes #9059 

Coauthored by @sakibh 

# Changelog

- add "get labware offset data" link to bottom of Labware Position Check explanation on the Labware Setup Step
- link launches closable modal with simple ui to browse and copy generated python snippets for CLI/Jupyter offset application 

# Review requests

- Upload a protoc [... cut]
commit c400c78d (2022-01-27): refactor(app): add advanced setting gated link to raw labware offset data for jupyter/cli execution (#9311)

Include generated labware load and offset snippets of python that can be copied and pasted into
jupyter or cli protocol execution. Link is only accessible behind an app advanced setting.

Closes #9059
closed 2022-01-27 by b-cooper