Opentrons/opentrons #9109: Complex protocol fails to load on beta.3

mattwelch, 2021-12-15:
5.0.0-beta.3

Complex protocol w/ custom labware (files attached) fails to load correctly on beta.3 (not tried on .4). This happens w/ both fast and slow sim.

mcous, 2021-12-15:
When testing with the dev server, this appears to be an app side issue. On my computer, the protocol can load and simulate properly, but when I click the "Run" tab, the UI effectively freezes. Eventually, the debug console in the app spits out:

```
react_devtools_backend.js:6 Warning: React instrumentation encountered an error: RangeError: Maximum call stack size exceeded
```

My hunch is that this is because the app sends out a `GET` for every single command in the run log upon run page load

SyntaxColoring, 2021-12-21:
Hypotheses on back-end causes: https://opentrons.slack.com/archives/C01F7KPNASV/p1639596222046500?thread_ts=1639577424.026500&cid=C01F7KPNASV

PR #9170 (2022-01-03): perf(protocol-engine): avoid iterating all commands for every state update
## Overview

This PR refactors the logic used to lookup the next queued command ID. Previously, this logic was O(n) in the worst case, and it was called by the engine's `QueueWorker` on every single state update. Legacy protocol execution triggered the worst case due to its bypass of the `queued` command state.

Closes #9109, closes #8739. Hopefully also has an impact on #9133.

## Changelog

- Maintain a separate list of queued command IDs to ensure queue lookup is always O(1)
- Maintain PE state in mutable dataclasses [... cut]
commit b12bffce (2022-01-03): perf(protocol-engine): avoid iterating all commands for every state update (#9170)

Closes #9109, closes #8739
closed 2022-01-03 by sanni-t