Opentrons/opentrons #9113: bug: Pipette tip crashes into usascientific_12_reservoir_22ml

danolson1, 2021-12-15:
# Overview
When using a single channel pipette with the usascientific_12_reservoir_22ml, the pipette crashes into the bottom. The problem is that  the wells are subdivided near the bottom, and pipetting in the exact center of the well causes the pipette to hit the divider between rows D and E

# Steps to reproduce
Aspirate from usascientific_12_reservoir_22ml with a single channel pipette at default (1 mm) pipette height above well bottom.

# Current behavior
Pipette tries to aspirate from in between the D1 and E1 positions of the reservoir, causing the tip to crash into the bottom.

# Expected behavior
The pipette should be offset slightly so that it aspirates from the D1 position, where the reservoir well is the expected depth

yinshiyi, 2021-12-15:
yo, do you have the python script?

danolson1, 2021-12-16:
Sure, here it is. The v1 file exhibits the crashing behavior. The v2 file
has a workaround, where I manually move the tip 4.5 mm in the y-direction.

v1 version
water = reservoir.wells()[0]

v2 version
water = reservoir.wells()[0].bottom().move(types.Point(0, 4.5, 1)) # offset
4.5 mm in the y direction to avoid collision

-Dan

# Remember to include imports in protocol file for the robot
from opentrons import protocol_api, simulate, types

metadata = {'apiLevel': '2.11',
            'author'  : 'Dan Olson'}

# set up constants
DEFAULT_WELL_BOTTOM_CLEARANCE = 1.0 # mm above well bottom
ABOVE_LIQUID_HEIGHT = 8.0 # mm above well bottom
STDCRV_START_WELL = 'A19'
STDCRV_REPLICATES = 1
STDCRV_DILUTIONS = 5
BLANK_WELLS = 0 # number of blanks to include at bottom of standard curve

# calculate starting indices for row and column of standard curve
stdcrv_rowidx = ord(STDCRV_START_WELL[:1].upper())-65 # convert first letter of well ID to a row index, starting from zero
stdcrv_colidx = int(STDCRV_START_WELL[1:]) - 1

# define the protocol in a function called run
def run(protocol: protocol_api.ProtocolContext):
    ## set up the machine
    protocol.comment('  ### SETTING UP THE INSTRUMENT ###') # adds a user-readable comment string
    temp_module = protocol.load_module('temperature module gen2', '7') # temperature module loaded into position 7 on the deck
    assay_plate = temp_module.load_labware('corning_384_wellplate_112ul_flat') # 384 well plated loaded onto the temperature module
    temp_module.set_temperature(4) # set the temperature module to 4C while loading the plate
    tube_rack = protocol.load_labware('opentrons_24_tuberack_eppendorf_1.5ml_safelock_snapcap', '4')
    reservoir = protocol.load_labware('usascientific_12_reservoir_22ml', '5') # 12 well reservoir loaded onto position 5
    tiprack_1 = protocol.load_labware('opentrons_96_tiprack_300ul', '1')
    tiprack_2 = protocol.load_labware('opentrons_96_tiprack_20ul', '2')
    p300m = protocol.load_instrument('p300_multi_gen2', 'left', tip_racks=[tiprack_1])
    p20 = protocol.load_instrument('p20_single_gen2', 'right', tip_racks=[tiprack_2])

    ## set up liquids
    water = reservoir.wells()[0].bottom().move(types.Point(0, 4.5, 1)) # offset 4.5 mm in the y direction to avoid collision
    buffer = reservoir.wells()[1]
    nad_1 = tube_rack.wells()[0]
    nad_2 = tube_rack.wells()[1]
    nad_3 = tube_rack.wells()[2]
    nad_4 = tube_rack.wells()[3]
    formate_1 = tube_rack.wells()[4]
    formate_2 = tube_rack.wells()[5]
    formate_3 = tube_rack.wells()[6]
    formate_4 = tube_rack.wells()[7]
    nadh = tube_rack.wells()[8]
    trash = protocol.fixed_trash['A1']
    
    ## set up the dilutions (do this by hand for now, might automate later)
    
    ## set up the standard curve in 20 ul volume so we can make the dilution series with the p20 pipette
    # load 20 ul water into wells B23:M24. The last 2 wells in each column are for blanks
    protocol.comment('  ### SETTING UP STANDARD CURVE ###')
    protocol.comment('      Adding 20 ul water')
    
    # define the wells
    stdcrv_wells = [] # empty list to hold wells
    for rep in range(STDCRV_REPLICATES):
        stdcrv_wells.extend(assay_plate.columns()[stdcrv_colidx+rep][stdcrv_rowidx+1:STDCRV_DILUTIONS+BLANK_WELLS]) # the +1 is to exclude the first well
    # add water to all wells, except first ones
[... 88 lines cut ...]
    # add 10 ul water to bring the final volume up to 30 ul
    protocol.comment('      Bringing up final volume to 30 ul with water, pipette above liquid level')
    p20.well_bottom_clearance.dispense = ABOVE_LIQUID_HEIGHT # dispense the last 10 ul above the liquid level
    protocol.comment(f'      Dispense height = {p20.well_bottom_clearance.dispense}')
    p20.distribute(10, 
                   water, 
                   assay_plate.wells()[352:364] + assay_plate.wells()[368:380],
                   touch_tip = True,
                   disposal_volume = 0
                  )
    p20.well_bottom_clearance.dispense = DEFAULT_WELL_BOTTOM_CLEARANCE # reset the well bottom clearance
    protocol.comment(f'      Dispense height = {p20.well_bottom_clearance.dispense}')
    
    
    
    # load the assay wells
    protocol.comment('  ### LOADING ASSAY WELLS ###')
    
    # pause protocol, since the next steps are time-sensitive
    protocol.set_rail_lights(True) # turn the lights on to indicate that we're ready for the next step
    
    # bring the temperature up to 37C
    temp_module.set_temperature(37)
    
    # start the assay by loading 30 ul buffer + enzyme into each assay well and standard curve wells

eirinnm, 2022-03-30:
I can confirm this issue also occurs with `nest_1_reservoir_195ml`.

The coordinates for well A1 of this reservoir are:
https://github.com/Opentrons/opentrons/blob/edge/shared-data/labware/definitions/2/nest_1_reservoir_195ml/1.json

      "x": 63.88,
      "y": 42.74,
      "z": 4.55

Whereas for the nest_12_reservoir_15ml, where this problem doesn't occur, well A5 is:
https://github.com/Opentrons/opentrons/blob/edge/shared-data/labware/definitions/2/nest_12_reservoir_15ml/1.json

      "x": 50.38,
      "y": 42.78,
      "z": 4.55

Changing the X coordinate for the 1_well_reservoir from 63.88 to 50.38 should fix this problem.

SyntaxColoring, 2024-08-14:
Hey everyone, thanks for bringing this to our attention and sorry we seem to have missed this thread until now.

We think @danolson1's workaround of adding a 4.5mm offset to the protocol is the best option for now. 

@eirinnm's solution of editing the labware definition might happen to work, but those dimensions are really supposed to represent the overall physical dimensions of the well. We try not to add one-off adjustments to get the pipette to go to the "right place" because it can be confusing and non-generalizable. I don't think that modified definition would do the right thing if you used it with the Flex's 96-channel pipette, for example.

closed 2024-08-14 by SyntaxColoring