Opentrons/opentrons #9125: Protocol Engine (PAPIv3): Pause when front door is opened

mattwelch, 2021-12-16:
Implement the existing pause-on-door-open feature for protocols that are run through Protocol Engine and the HTTP API. This work is limited to PAPIv3 protocols run via HTTP endpoints.

mcous, 2022-06-17:
Ticket under investigation as potential 6.1.0 release requirement

mcous, 2022-06-17:
This ticket is required for the 6.1.0 release. Updated tracking information accordingly

mcous, 2022-06-28:
After #10882 is merged, this is my implementation proposal to bring this home:

- Rename `HardwareEventForwarder` to `DoorWatcher`
- Simplify `HardwareEventAction` to `DoorChangeAction`
    - We don't use any other hardware events
    - When we do in the future, I don't want to continue to use this callback interface; it's too dangerous with the `ThreadManager` and protocol child thread in place
- On door change notification from the hardware control API:
    1. Check if queue status is `running`
    2. If it is, pause the hardware API
    3. Dispatch a `DoorChangeAction` with the door state
- Remove remaining action handling from the `LegacyContextPlugin`, and perhaps remove the abstract `handle_action` method from the generic plugin interface, too

PR #11021 (2022-07-11): feat(engine): pause protocol execution if door is opened
# Overview
Closes #9125 

Adds pausing protocol execution if a door state hardware notification is detected to protocol engine. Protocol Engine hardware action and handlers renamed to reflect dealing with door events only, along with simplification of the door change action.

# Changelog
- Pauses protocol execution upon `DoorStateNotification`
- Refactor and rename PE HardwareEvent names
- Remove legacy action handling

# Review requests
# Risk assessment
Low, mostly refactoring and renaming with only small and contained functionality ad [... cut]
closed 2022-07-11 by jbleon95