Opentrons/opentrons #9133: Protocol runs very slowly on 5.0.0-beta.4

mattwelch, 2021-12-17:
From a user:

"It was extremely slow. I ran a protocol involving a bunch of mixing and transferring and the robot was frequently pausing and much less fluid than usual. This was also seen on the app interface where I could not scroll through the protocol and when I hit pause there was sometimes up to a 25 second delay. I am hoping this is just related to being in a beta version and not do to changes in the app."

See Matt for a copy of the protocol

SyntaxColoring, 2021-12-17:
I bet this is the problem we theorized in https://github.com/Opentrons/opentrons/pull/8701#pullrequestreview-803063214. I've sent the user a troubleshooting step to confirm.

SyntaxColoring, 2022-01-20:
This will be closed by the combination of:

* #9170
  Access certain data much more efficiently while occupying the event loop, so we don't hog the event loop as much.
* #9199
  Make it so hardware commands don't have to wait for the main thread's event loop to be free.
* #9238
  Make it so the protocol thread doesn't have to wait for the main thread's event loop to be free when the protocol thread has activity to report.
* #9188
  Make response serialization faster and avoid some unnecessary validation.
* #9197
  Move response serialization out of the event loop and into a worker thread. 

Unfortunately, despite all of this, there's still a user-facing problem of the robot moving more slowly and jittery when it gets busy with HTTP requests. 😢  I will ticket this separately.

closed 2022-01-26 by SyntaxColoring