Opentrons/opentrons #9141: bug: Jerky motion during touch_tip() command

danolson1, 2021-12-17:
# Overview
When I have the robot do a touch_tip() operation, it immediately jerks the tip before starting the touch_tip() operation. I think this might be causing small droplets of liquid to be sprayed into adjacent wells.

# Steps to reproduce
Run the touch_tip() command on a 384 well plate at a slow speed (5 mm/min). I'm sure it happens at fast speeds too, but it is much more visually obvious when set to a slower speed.

I am doing this via the Jupyter interface, in case that matters.

# Current behavior
The tip immediately jerks back and forth, then proceeds to slowly touch the walls.

# Expected behavior
No jerky behavior. This might require smoothing the acceleration of the motion commands.

danolson1, 2021-12-18:
When I set the touch_tip parameter to True in the dispense() function, and use it on a full column of a 384 well plate, the jerky motion seems to be causing the Y-axis to drift, which eventually causes pipetting in the wrong wells.

SyntaxColoring, 2021-12-20:
@danolson1 Which robot software version is this on?

danolson1, 2021-12-20:
Server version 4.7.0
Firmware version: v1.1.0-25e5cea

SyntaxColoring, 2021-12-21:
Got it. I ask because our *v5.0 beta* is having problems where the whole `touch_tip()` is slower (#9135), presumably because it's erroneously stalling for a moment between each movement within the `touch_tip()`.

But it sounds like you're talking about something different: the movements are coming right after each other, as they should, but the change in direction is too abrupt?