Opentrons/opentrons #9150: bug: Jupyter looks for deck_calibration.json and robot_setting.json in the wrong places

danolson1, 2021-12-20:
# Overview
When I use the Jupyter interface, when I set up a simulation protocol, the robot looks for the configuration files in the wrong place and gives me an error message.

# Steps to reproduce
Open the Jupyter interface

```
from opentrons import simulate
protocol = simulate.get_protocol_api('2.11')
```

# Current behavior
The following error messages are generated
/data/deck_calibration.json not found. Loading defaults
/data/robot_settings.json not found. Loading defaults

# Expected behavior
The deck_calibration.json file is actually stored in /data/robot/deck_calibration.json
I can't find the robot_settings.json file. Maybe it's the one currently called config.json?

Make sure that the simulate module is looking for the files in the right location. Since these calibration files are actually not that important when doing a simulation, you might consider suppressing the error messages, or noting that they're for the simulate function and can safely be ignored.

danolson1, 2021-12-21:
I get this error when running the `opentrons.execute.get_protocol_api('2.11')` command as well.

Here's the full text of the error message:
/data/robot_settings.json not found. Loading defaults
`Failed to initialize character device, will not be able to control gpios (lights, button, smoothiekill, smoothie reset). Only one connection can be made to the gpios at a time. If you need to control gpios, first stop the robot server with systemctl stop opentrons-robot-server. Until you restart the server with systemctl start opentrons-robot-server, you will be unable to control the robot using the Opentrons app.`

Matt-Zwimpfer, 2022-06-02:
@hartm9 might want to bring this back up at some point (found during Case Review revamp)

mcous, 2022-07-27:
> /data/deck_calibration.json not found. Loading defaults
> /data/robot_settings.json not found. Loading defaults

These are overly chatty logs; these are very old locations for where these files used to live, so `not found` is entirely expected, and in the background the system is finding and using the newer, correct locations.

> Failed to initialize character device, will not be able to control gpios (lights, button, smoothiekill, smoothie reset). Only one connection can be made to the gpios at a time. If you need to [...]

This is an expected message if the device's HTTP server is running while you use Jupyter. The "GPIOs" that the message is referring to are the OT-2's lights and buttons. If you do not need to use these devices in your protocol, you can ignore these message. If you _do_ need to control lights via Jupyter, you should follow the instructions in the message and issue a `systemctl start opentrons-robot-server` via the terminal to stop the robot's HTTP server.

closed 2022-07-27 by mcous