Opentrons/opentrons #9163: Protocol upload - Protocol name is not updating

mikecardarelli, 2021-12-23:


sakibh, 2022-01-03:
I am able to see the protocol names for both JSON and Python protocols (screenshots below) on the latest version (5.0.0-beta.5). Perhaps the JSON protocol was exported without a title so it displayed as "Untitled"?

[image]

[image]

b-cooper, 2022-01-10:
Investigate if the name is properly falling back to the filename

mikecardarelli, 2022-01-11:
Anthony W. to attempt to repro on the new BETA

anthonywilleee, 2022-01-12:
@b-cooper I gave this shot on 5.0.0-beta.5

Sakib is correct - if the JSON protocol is exported without a protocol title the name reads "untitled"
If the JSON protocol is exported with a title, the title displays.

If that's the intended functionality I think we're good here.  FWIW, it's easy to miss the title field in PD, so I expected the filename to show as the protocol name.

anthonywilleee, 2022-01-12:
@b-cooper followed up with Emily on this, shes agrees something we can tackle later, perhaps with milestone 1 work. Closing this ticket.

mikecardarelli, 2022-01-20:
[https://app.zenhub.com/files/38644841/a584c9a3-0357-4bf3-b547-21c86974f1e9/download](https://app.zenhub.com/files/38644841/a584c9a3-0357-4bf3-b547-21c86974f1e9/download)

PR #9307 (2022-01-26): fix(app): add fileName to title bar if protocolName is undefined
closes #9163

# Overview

Currently, if you have a protocol with no name, the title bar at the top of the `RunDetails` and `ProtocolUpload` page doesn't include any protocol name. Now, if you upload a protocol with no name, it will include the `fileName` in the title bar.

Here is how it looks from `protocolUpload` page:
[image]

And from the `runDetails` page
[image]

# Changelog

- both `runDetails` and `protocolUpload` grab the `fileName` from protocol resource

# Review requests

- test on your robot, you should now see the f [... cut]
commit 18cc3acd (2022-01-26): fix(app): add fileName to title bar if protocolName is undefined (#9307)

closes #9163
closed 2022-01-26 by jerader