Opentrons/opentrons #9175: bug: Thermocycler fails performing basic PCR program after lid closing/shifting temperatures

peradastra, 2022-01-01:
# Overview
When running a simple PCR program the thermocycler (Tc) fails with a uniformity error after lid closing. Interestingly, this error **only** happens when the Tc has been turned off long enough for the lid and plate to cool down to the ambient temperature. 

# Steps to reproduce
0) Turn off all hardware (OT2* & Tc**) and allow Tc plate and lid to return to ambient (~23°C) temperature. 
1) Turn on OT2.
2) Turn on Tc.
3) Upload [this](https://www.dropbox.com/s/l272q803jodjygw/thermocycler_testing.py?dl=0) run protocol. 
4) After simulation is complete, initiate protocol.

**OT2 server version is 4.7.0.* \
**Tc firmware version is v1.1.0.*

# Documentation of error
I was able to reproduce this error 3 separate times using 2 different OT2s (A & B) and 3 different Tc's (A, B, & C). Below are 3 videos simultaneously showing the OT2 and app running in real-time up until they encounter the error. Below is also one API log which I downloaded and parsed only for the error below.  

### Videos
[image]

[image]

[image]

### API Logs

    Dec 31 22:08:33 e9eb27 opentrons-api[208]: rpc built call run on <opentrons.api.session.Session object at 0x70c47690>
    Dec 31 22:08:33 e9eb27 opentrons-api[208]: Updating instrument model cache
    Dec 31 22:08:34 e9eb27 opentrons-api[208]: loaded: p300_multi_v2.0, pipette offset: [-1.5000000000000089, -1.3000000000000185, 1.5987211554602254e-14]
    Dec 31 22:08:34 e9eb27 opentrons-api[208]: Doing full configuration on LEFT
    Dec 31 22:08:34 e9eb27 opentrons-api[208]: Updated move split config with {'B': MoveSplit(split_distance=1, split_current=1.75, split_speed=1, after_time=1800, fullstep=True)}
    Dec 31 22:08:34 e9eb27 opentrons-api[208]: loaded: p20_multi_v2.1, pipette offset: [1.6000000000000014, -2.3000000000000043, 0.7999999999999972]
    Dec 31 22:08:34 e9eb27 opentrons-api[208]: Doing full configuration on RIGHT
    Dec 31 22:08:34 e9eb27 opentrons-api[208]: Instruments found: {<Mount.LEFT: 1>: <Pipette: P300 8-Channel GEN2 1892634800>, <Mount.RIGHT: 2>: <Pipette: P20 8-Channel GEN2 1873690832>}
    Dec 31 22:08:34 e9eb27 opentrons-api[208]: Starting hardware event watcher
    Dec 31 22:08:34 e9eb27 opentrons-api[208]: Session.set_state state=running user_message=None duration=None reason=None
    Dec 31 22:08:34 e9eb27 opentrons-api[208]: Session.set_state state=running user_message=None duration=None reason=None
    Dec 31 22:08:34 e9eb27 opentrons-api[208]: Homing axes ZA in sequence ['ZA']
    Dec 31 22:08:36 e9eb27 opentrons-api[208]: Updating instrument model cache
    Dec 31 22:08:38 e9eb27 opentrons-api[208]: Skipping configuration on LEFT
    Dec 31 22:08:38 e9eb27 opentrons-api[208]: Skipping configuration on RIGHT
    Dec 31 22:08:38 e9eb27 opentrons-api[208]: Instruments found: {<Mount.LEFT: 1>: <Pipette: P300 8-Channel GEN2 1892634800>, <Mount.RIGHT: 2>: <Pipette: P20 8-Channel GEN2 1873690832>}
    Dec 31 22:08:38 e9eb27 opentrons-api[208]: Resetting configuration for LEFT
    Dec 31 22:08:38 e9eb27 opentrons-api[208]: loaded: p300_multi_v2.0, pipette offset: [-1.5000000000000089, -1.3000000000000185, 1.5987211554602254e-14]
    Dec 31 22:08:38 e9eb27 opentrons-api[208]: Resetting configuration for RIGHT
    Dec 31 22:08:38 e9eb27 opentrons-api[208]: loaded: p20_multi_v2.1, pipette offset: [1.6000000000000014, -2.3000000000000043, 0.7999999999999972]
    Dec 31 22:08:38 e9eb27 opentrons-api[208]: /data/robot_settings.json not found. Loading defaults
    Dec 31 22:08:38 e9eb27 opentrons-api[208]: /data/deck_calibration.json not found. Loading defaults
    Dec 31 22:08:38 e9eb27 opentrons-api[208]: Updating instrument model cache
    Dec 31 22:08:38 e9eb27 opentrons-api[208]: Skipping configuration on LEFT
    Dec 31 22:08:38 e9eb27 opentrons-api[208]: Skipping configuration on RIGHT
    Dec 31 22:08:38 e9eb27 opentrons-api[208]: Instruments found: {<Mount.LEFT: 1>: None, <Mount.RIGHT: 2>: None}
    Dec 31 22:08:38 e9eb27 opentrons-api[208]: Updating instrument model cache
    Dec 31 22:08:39 e9eb27 opentrons-api[208]: Skipping configuration on LEFT
    Dec 31 22:08:39 e9eb27 opentrons-api[208]: Skipping configuration on RIGHT
    Dec 31 22:08:39 e9eb27 opentrons-api[208]: Instruments found: {<Mount.LEFT: 1>: <Pipette: P300 8-Channel GEN2 1873692752>, <Mount.RIGHT: 2>: <Pipette: P20 8-Channel GEN2 1892636080>}
    Dec 31 22:08:39 e9eb27 opentrons-api[208]: Homing axes XYZAXZA in sequence ['ZA', 'X', 'Y']
    Dec 31 22:08:45 e9eb27 opentrons-api[208]: Homing axes B in sequence ['B']
    Dec 31 22:08:52 e9eb27 opentrons-api[208]: Homing axes C in sequence ['C']
    Dec 31 22:08:58 e9eb27 opentrons-api[208]: command.THERMOCYCLER_OPEN: 
    Dec 31 22:08:58 e9eb27 opentrons-api[208]: Cannot assure a safe gantry position to avoid colliding with the lid of the Thermocycler Module.
    Dec 31 22:08:58 e9eb27 opentrons-api[208]: command.THERMOCYCLER_SET_BLOCK_TEMP: temperature: 4, hold_time: None
    Dec 31 22:09:39 e9eb27 opentrons-api[208]: command.THERMOCYCLER_SET_LID_TEMP: 
    Dec 31 22:12:17 e9eb27 opentrons-api[208]: command.THERMOCYCLER_CLOSE: 
    Dec 31 22:12:17 e9eb27 opentrons-api[208]: Cannot assure a safe gantry position to avoid colliding with the lid of the Thermocycler Module.
    Dec 31 22:12:43 e9eb27 opentrons-api[208]: command.THERMOCYCLER_OPEN: 
    Dec 31 22:12:43 e9eb27 opentrons-api[208]: Cannot assure a safe gantry position to avoid colliding with the lid of the Thermocycler Module.
    Dec 31 22:13:12 e9eb27 opentrons-api[208]: command.THERMOCYCLER_CLOSE: 
    Dec 31 22:13:12 e9eb27 opentrons-api[208]: Cannot assure a safe gantry position to avoid colliding with the lid of the Thermocycler Module.
    Dec 31 22:13:36 e9eb27 opentrons-api[208]: Thermocycler has encountered an unrecoverable error: /dev/ot_module_thermocycler0: 'Received error response 'ERROR:Plate temperature not uniform. Deactivating.
                                               Error:Plate temperature is not uniform
[... 61 lines cut ...]
                                               Lid:closed'
                                               
                                               The above exception was the direct cause of the following exception:
                                               
                                               Traceback (most recent call last):
                                                 File "usr/lib/python3.7/site-packages/robot_server/service/legacy/rpc/rpc.py", line 267, in make_call
                                                 File "usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
                                                 File "usr/lib/python3.7/site-packages/robot_server/service/legacy/rpc/rpc.py", line 258, in call_and_serialize
                                                 File "usr/lib/python3.7/site-packages/opentrons/api/session.py", line 657, in run
                                                 File "usr/lib/python3.7/site-packages/opentrons/api/util.py", line 33, in decorated
                                                 File "usr/lib/python3.7/site-packages/opentrons/api/session.py", line 649, in _run
                                                 File "usr/lib/python3.7/site-packages/opentrons/api/session.py", line 627, in _run
                                                 File "usr/lib/python3.7/site-packages/opentrons/protocols/execution/execute.py", line 27, in run_protocol
                                                 File "usr/lib/python3.7/site-packages/opentrons/protocols/execution/execute_python.py", line 71, in run_python
                                               opentrons.protocols.execution.errors.ExceptionInProtocolError: ErrorResponse [line 20]: /dev/ot_module_thermocycler0: 'Received error response 'ERROR:Plate temperature not uniform. Deactivating.
                                               Error:Plate temperature is not uniform
                                               Lid:closed'
# Current behavior

I suspect that this behavior is the result of one, or both, already notes issues. Ticket [#5602](https://github.com/Opentrons/opentrons/issues/5602) documents this exact error where a plate uniformity error arises if the lid is opened or closed immediately after `set_block_temperature()`. 
[... cut ...] ly for PCR. I would suspect that the acceptable TEMP_THRESHOLD` calculated from extended at a specific temperature would lead to such a small value that cycling out of that temperature would return a plate uniformity error only unless the Tc did not get to the next temperature quickly and without overshooting the target temperature.

# Expected behavior

For the thermocycler to be able to not fail when doing basic PCR protocols and not when shifting from 1 long incubation to another temperature.

peradastra, 2022-01-02:
# Update
I built a new version of the Tc firmware locally from the opentrons-module repo which includes commits that allow the print out the thermistors values and uploaded it to the Tc in the event of failure. Here they are...

ErrorResponse [line 43]: /dev/ot_module_thermocycler2: 'Received error response 'Error:Plate temperature is not uniform. T1: 2.0172	T2: 8.0703	T3: 4.2907	T4: 3.7157	T5: 8.7126	T6: 4.4932	T.Lid: 105.6111	T.sink: 61.8712	T_error: 0.0000

sanni-t, 2022-02-04:
Thank you for the detailed recording and troubleshooting steps of this issue!

The behavior you are seeing is not due to having the thermocycler turned off for a long time but due to the consecutive, multiple lid open & close commands (although, having the module start off from an ambient temperature could be aggravating the issue slightly). The issue lies in the way the lid's movement affects internal temperature reading. This behavior is documented in https://github.com/Opentrons/opentrons-modules/issues/104.

tl;dr of the issue is that every time the lid is moving, the thermocycler is unable to intelligently correct its block temperature. So each time the lid is opened or closed, the block/plate's temperature deviates slightly. It gets corrected immediately after the motion is completed. But if you have consecutive open and close commands, the thermocycler doesn't get any time to correct its plate's temperature; each consecutive lid command adds to the temperature deviation, ultimately making it large enough to result in a uniformity error.

sanni-t, 2022-02-04:
> While not documented in this ticket, I have a similar issue where the Tc returns a plate uniformity error after holding at a temperature for 15 minutes and then shifts to another temperate afterward. 

This seems to be a different and more concerning issue. Would you mind creating a different ticket for it with the observed behavior and logs? Did this protocol also have consecutive lid open and close commands?

> For example, an RT-PCR reaction is first done by incubation at a specific temperature for several minutes followed by cycling through several different temperatures quickly for PCR. I would suspect [...]

The `TEMP_THRESHOLD` value you are referring to does not detect the uniformity of the plate. This value is only monitored for a short amount of time in the python API to check whether the temperature set after `set_temperature` has stabilized, *before* moving on to the next protocol step. The uniformity detection is done by the thermocycler firmware which continuously monitors the temperatures of different spots on the plate/block. It raises the error only when the plate is supposed to have reached stable temperatures and two spots on the plate are at very different temperatures, like those shown in your logs `T1: 2.0172` and `T2: 8.0703`

peradastra, 2022-02-11:
Thanks for the thorough reply and all your hard work in particular addressing the Tc issues! I've been following your commits as I tried to debug. I'll try to adjust my protocols accordingly. I've been automating longer protocols which require multiple opening and closings of the Tc but will break them down. Though once this initially error happens and the Tc is "warmed up," I don't see this error happen again. 

> Thank you for the detailed recording and troubleshooting steps of this issue!
> 
> The behavior you are seeing is not due to having the thermocycler turned off for a long time but due to the consecutive, multiple lid open & close commands (although, having the module start off from an ambient temperature could be aggravating the issue slightly). The issue lies in the way the lid's movement affects internal temperature reading. This behavior is documented in [Opentrons/opentrons-modules#104](https://github.com/Opentrons/opentrons-modules/issues/104).
> 
> tl;dr of the issue is that every time the lid is moving, the thermocycler is unable to intelligently correct its block temperature. So each time the lid is opened or closed, the block/plate's temperature deviates slightly. It gets corrected immediately after the motion is completed. But if you have consecutive open and close commands, the thermocycler doesn't get any time to correct its plate's temperature; each consecutive lid command adds to the temperature deviation, ultimately making it large enough to result in a uniformity error.
> 
> We don't have a fix planned for this behavior immediately as having the lid open and close multiple times consecutively is not something a protocol would typically need to do. That said, improving the lid functionality is something on our radar.

peradastra, 2022-02-11:
> > While not documented in this ticket, I have a similar issue where the Tc returns a plate uniformity error after holding at a temperature for 15 minutes and then shifts to another temperate afterward.
> 
> This seems to be a different and more concerning issue. Would you mind creating a different ticket for it with the observed behavior and logs? Did this protocol also have consecutive lid open and close commands?
> 
> > For example, an RT-PCR reaction is first done by incubation at a specific temperature for several minutes followed by cycling through several different temperatures quickly for PCR. I would suspect that the acceptable TEMP_THRESHOLD` calculated from extended at a specific temperature would lead to such a small value that cycling out of that temperature would return a plate uniformity error only unless the Tc did not get to the next temperature quickly and without overshooting the target temperature.
> 
> The `TEMP_THRESHOLD` value you are referring to does not detect the uniformity of the plate. This value is only monitored for a short amount of time in the python API to check whether the temperature set after `set_temperature` has stabilized, _before_ moving on to the next protocol step. The uniformity detection is done by the thermocycler firmware which continuously monitors the temperatures of different spots on the plate/block. It raises the error only when the plate is supposed to have reached stable temperatures and two spots on the plate are at very different temperatures, like those shown in your logs `T1: 2.0172` and `T2: 8.0703`. Merely cycling through different temperatures shouldn't result in this error. So it'll be helpful for us to see the protocol you are using and where this error is happening.

Yeah, I'll open a new ticket this week documenting the above mentioned error modality. Thanks again for you feedback on this.