Opentrons/opentrons #9187: bug: OT-2 continues with pipetting steps while temperature module is still completing a .set_temperature() step

Matt-Zwimpfer, 2022-01-05:
# Overview

While running the following code on software and robot server version 4.5.0:

`temp_module.set_temperature(22)
temp_module.set_temperature(67)
for _ in range(10):
    pipette_l.pick_up_tip()
    pipette_l.drop_tip()`
  
observed that robot waits for the temperature module to reach 22C, but then as the module starts ramping up to 67C, the robot  simultaneously proceeds with picking up and dropping off tips. The robot should wait for the module to reach 67C before going through the rest of the protocol. 

# Steps to reproduce
1. Create a Python API protocol with the above code/steps 
2. Upload to an OT-2  running 4.5.0 (though expect this still persists up to the latest software version)
3. Run protocol

# Current behavior
As the second `.set_temperature()` command commences, the robot starts running the rest of the code, in this case the `for` loop for picking up and dropping tips. 

# Expected behavior
The robot should wait for the `.set_temperature()` to complete with the temperature module reaching the programmed temperature before running through the rest of the protocol.

amitlissack, 2022-01-05:
A cursory look at the code indicates that @sanni-t may have fixed this in September. I will double check.

Matt-Zwimpfer, 2022-01-06:
Thanks @amitlissack! I had looked through the existing and closed` set_temperature()` bugs, but couldn't find a ticket describing this specific problem.

amitlissack, 2022-01-07:
https://github.com/Opentrons/opentrons/issues/8275

sanni-t, 2022-01-07:
^ Yep. That should have fixed it. Thanks @amitlissack! 
@Matt-Zwimpfer does this error still show up in 4.7?

Matt-Zwimpfer, 2022-01-07:
Ah thanks @amitlissack and @sanni-t ! Thought the #8275 fix for `await_temperature()` was separate from this. The user that brought this up is still on 4.5.0, so that is most likely the cause. 

Should I delete this ticket?

amitlissack, 2022-01-07:
I should have noticed when Ethan first posted this that the customer is doing two `set_temperature` calls consecutively.

You can safely delete this.