Opentrons/opentrons #9253: bug: Depth-first search relies on undefined set iteration order

SyntaxColoring, 2022-01-14:
https://github.com/Opentrons/opentrons/blob/09a83ed868a3c73ced2593ea6407e6c8ba92de42/api/src/opentrons/algorithms/dfs.py#L37-L50

This says it's meant to return "the set of visited vertices **in depth first search order,**" but `set` is unordered. See [these official Python docs](https://docs.python.org/3/library/stdtypes.html#set-types-set-frozenset), [and this Stack Overflow question](https://stackoverflow.com/questions/9792664/converting-a-list-to-a-set-changes-element-order).

mcous, 2022-01-20:
I believe MoaM relies on this code. Anecdotally, sets being unordered is a thing that has tripped me up. I feel like this is a fairly risky bug and I'm going to mark is as 5.0 accordingly

sfoster1, 2022-02-09:
The return comment may not be right but it also shouldn't matter what order these values are returned in as long as that order is consistent with the same input.

PR #9620 (2022-03-07): fix(api): sort modules in ascending port order
## Overview

This PR takes over for #9412.

[Our documentation](https://docs.opentrons.com/v2/new_modules.html#using-multiple-modules-of-the-same-type) states that modules are listed/loaded from "left to right". In hardware terms, that seems to mean lower ports to higher ports.

Prior to this PR, the API sorted modules in descending port order, resulting in modules loading from right to left. Closes #9309.

## Changelog

- Move module sorting out of `USBBus` class and into `ModuleControl` class
    - This has the pleasant effect of **sorting the G [... cut]
commit 4e65e988 (2022-03-07): fix(api): sort modules in ascending port order (#9620)

Closes #9309, closes #9253
closed 2022-03-07 by mcous