Opentrons/opentrons #9293: bug: Routine polls from the Opentrons App consume the OT-2's CPU bandwidth and make the OT-2's movement slow and jittery

SyntaxColoring, 2022-01-21:
# Overview

When running a protocol, the OT-2s movements get unusually janky (jittery, stuttery, sluggish, slow) while the robot is handling routine network requests from the Opentrons App. It appears to be because handling the request consumes a lot of the robot's CPU bandwidth.

# Steps to reproduce

Upload and run this protocol on a v5.0 beta app and robot:

```python
metadata = {
    "apiLevel": "2.0",
}

REPEAT_COUNT = 500

def run(context):
    tip_rack = context.load_labware("opentrons_96_tiprack_300ul", 1)
    labware = context.load_labware("biorad_96_wellplate_200ul_pcr", 2)
    pipette = context.load_instrument("p300_multi_gen2", "left", [tip_rack])

    pipette.pick_up_tip()
    pipette.move_to(labware.wells()[0].top())
    for i in range(REPEAT_COUNT):
        pipette.touch_tip()
    pipette.return_tip()
```

You'll occasionally see the robot's movement stutter between the individual movements within the `touch_tip()`.

The problem will go away when you close the Opentrons App, and come back when you reopen it.

The problem gets worse when:

* The protocol does many small hardware movements instead of a few large ones.
* Network responses get bigger, more complex, or more frequent. For example, there's a spurt of stuttering when you close and reopen the app and the app sends out a bunch of `GET /runs/{id}/commands/{id}` requests to catch up. And I haven't tested this, but I'm also pretty sure that if you left this protocol running, the problem will gradually get worse over time until the protocol ends, as the response to `GET /runs/{id}` (which the app polls regularly) gets larger and larger.

This Bash command will artificially spam requests so you can see the problem more easily:

```bash
while true; do; curl --location --request GET 'my-robot-ip:31950/protocols' --header 'Opentrons-Version: 2' > /dev/null 2>&1 && date; done
```

# Acceptance criteria

This issue can be closed when normal polls from the Opentrons App do not cause noticeable stuttering.

For these purposes, long protocols with thousands or tens of thousands of steps should count as "normal."

# Technical investigation

The related issue #9133 addressed a different underlying technical cause of these same symptoms. The theory in *that* issue was that various things were hogging our async event loop, and the protocol's smooth progress was unnecessarily dependent on that event loop staying free.

However, symptoms persist despite solving #9133. This suggests that we also have a problem with network requests using up all our *CPU bandwidth,*  not just hogging our async event loop. 

As a soft confirmation of this theory, you can modify `get_protocols()` like this:

```python
    # Hog the event loop for at least 10 seconds.
    import time
    time.sleep(10)

    return await PydanticResponse.create(
        content=SimpleMultiBody.construct(data=data),
        status_code=status.HTTP_200_OK,
    )
```

If you then hit `GET /protocols` while the protocol is running, the robot will only be slow and stuttery for the ~1 second it takes to serialize the response, *not* the ~11 seconds it occupies the event loop. This shows that the problem *isn't* still that something is hogging the event loop.

The same thing happens if you modify `PydanticResponse.create()` like this:

```python
        def do_in_thread() -> "PydanticResponse":
            # Hog a worker thread for at least 10 seconds.
            time.sleep(10)
            return cls(content, status_code)

        return await to_thread.run_sync(do_in_thread)
```

This shows that the problem *isn't* that we're sharing a thread pool with something lower-level, and hogging a thread, preventing that lower-level thing from acquiring it for itself.

Finally, if you modify `app_setup.on_startup()` like this:

```python
@app.on_event("startup")
async def on_startup() -> None:
    """Handle app startup."""
    initialize_logging()
    initialize_hardware(app.state)

    import threading
    def busy_work() -> None:
        x = 0
        while True:
            x = x**2
    thread = threading.Thread(target=busy_work, name="Busy work", daemon=True)
    thread.start()
```

Then the protocol run will be consistently slow and stuttery regardless of network activity. This shows that CPU busyness can cause these user-facing symptoms.

SyntaxColoring, 2022-01-21:
# Rough ideas for solutions

* Adjust thread priority. What happens if we tell the OS scheduler to prioritize the protocol thread and hardware communication thread over our serialization worker threads?
* Tune the interval at which the OS switches between threads.
  * I presume the protocol thread is doing a very small amount of work compared to the serialization thread. A more granular interval could give the protocol thread more opportunities to intercede.
* Parallelism across more cores. Because of Python's GIL, we're currently pinning just one of our four cores at 100%.
* An HTTP API supporting lighter-weight polls.
* An HTTP API supporting less frequent polls.
* Faster JSON serialization.

mcous, 2022-03-25:
Closing in favor already completed work to ensure good (enough) behavior with the Opentrons App. Future platform concerns and actual resolution of the root issues moved to #9786

closed 2022-03-25 by mcous