Opentrons/opentrons #9309: bug: Multiple modules of the same type are swapped between protocol setup and execution

SyntaxColoring, 2022-01-25:


SyntaxColoring, 2022-01-26:
Possibly caused by #9253.

mcous, 2022-03-01:
Investigation results on v5.0.1:

- PAPIv2 / ProtocolEngine
    - Modules are "consumed" in descending order by port number
- App
    - Modules are assumed to be "consumed" in ascending order by port number
- App instructions
    - Users are instructed to plug modules in according to deck slot

Conclusions:

- The app and the API disagree on how load order maps to physical port order
- The app's instructions are straight up wrong and I can't think of any way we could make the system behave as the instructions specify

closed 2022-03-07 by mcous

splaisan, 2022-05-09:
we have a similar issue with two mag-modules but the issue is semi-random and occurs or not in the same protocol at a given code locations (works in the beginning of the protocol and swaps the deviced further on)
I suggest the program be improved to avoid having to sort anything, devices are declared on top of the code and should be addressed correctly I believe.
the fact our error is semi random suggests a more complex issue with USB ports being somehow wrongly addressed/incremented! ([ ref:_00Df44lpsm._5002S1SsrOb:ref ])

SyntaxColoring, 2022-05-09:
@splaisan Can you tell me what you see as:

1. **The version of your Opentrons App:** [image]
2. **Your OT-2's server version:** [image]

(In these example screenshots, mine is `5.1.0-beta.0`, but yours will probably say something different.)

The modules swapping themselves in the *middle* of a protocol is pretty alarming. Can you share your full protocol file along with any custom labware definitions that might be required to run it?

braldrich, 2022-05-09:
@SyntaxColoring I am working with this customer in Salesforce, but they are on 5.0.2 app and robot server version. 

@splaisan Please provide your protocol here or in our email correspondence.

splaisan, 2022-05-09:
Hi @braldrich 
I already provided the protocol with my above-linked ticket few days ago. and explained our issue in the ticket.
Can you access this directly? I am not at the lab now.

braldrich, 2022-05-09:
[Archive.zip](https://github.com/Opentrons/opentrons/files/8652983/Archive.zip)

Protocols here

PR #9620 (2022-03-07): fix(api): sort modules in ascending port order
## Overview

This PR takes over for #9412.

[Our documentation](https://docs.opentrons.com/v2/new_modules.html#using-multiple-modules-of-the-same-type) states that modules are listed/loaded from "left to right". In hardware terms, that seems to mean lower ports to higher ports.

Prior to this PR, the API sorted modules in descending port order, resulting in modules loading from right to left. Closes #9309.

## Changelog

- Move module sorting out of `USBBus` class and into `ModuleControl` class
    - This has the pleasant effect of **sorting the G [... cut]
commit 4e65e988 (2022-03-07): fix(api): sort modules in ascending port order (#9620)

Closes #9309, closes #9253