Opentrons/opentrons #9329: feat:

shlokamin, 2022-01-28:
# Overview

This ticket outlines the frontend side side of the heater shaker module card in device details.

# Implementation details
The module card should be composed out of a generic `ModuleCard` component that can be shared by all other modules

# Design
@howisthisnamenottakenyet please insert designs here when finalized

# Acceptance criteria
Module card should display...
- [ ] A rendering of the heater shaker module
- [ ] The port the module is connected in
- [ ] The heater shaker icon icon
- [ ] The module's display name
- [ ] The module's heater status (idle, heating, heated(??)) 
- [ ] The modules target temperature (in Celcius) 
- [ ] The module current temperature (in Celcius) 
- [ ] The module's shaker status (idle, shaking)
- [ ] The module's target RPM 
- [ ] The module's current RPM

anthonywilleee, 2022-01-28:
@shlokamin those are the correct heater statuses.  The dropdown menu may also have selection for set up and test shake, TBD based on user testing this week

anthonywilleee, 2022-02-09:
Dup with [9288](https://app.zenhub.com/workspaces/app--ui-5f6248dba845b000139aa767/issues/opentrons/opentrons/9288). Closing

closed 2022-02-09 by anthonywilleee