Opentrons/opentrons #9367: spike: Determine how many run logs/run sessions we should store

mattwelch, 2022-02-04:
We need to determine how many run logs we should store, and whether run logs are bundled up and saved with actual runs that are persisted in #8474.

## Acceptance criteria
- [ ] CPX determines reasonable number of logs to store, based on resource constrains and perceived likely log sizes
- [ ] CPX defines how logs are stored: part of past runs themselves, or separately.

mcous, 2022-02-10:
Our current UX designs account for 20 previous runs. We should validate the technical feasibility of this number

mcous, 2022-02-24:
Spike results:

- We expect 20 runs + protocols to be technically feasible
    - Backup plan if this doesn't pan out: drop down to 10
- What will we store?
    - Protocols
        - Full files, either on the filesystem or as database blobs
        - Full analysis results, which effectively means full command list
     - Runs
         - Labware offsets
         - Linkage to protocol
         - Full run log
             - This is a could candidate of something to drop if it turns out the UX doesn't need it

Next step is to finish #8496. Some SQLite investigation was done in the course of validating storage requirements, and so far it looks promising

closed 2022-02-24 by mcous