Opentrons/opentrons #9374: feat: Make a new endpoint submit and await jogging commands

mattwelch, 2022-02-04:
To improve what's perceived as a laggy jogging experience, we need to rethink how we implement jogging. Current thinking is a new endpoint with which to submit and await jogging commands.

SyntaxColoring, 2022-02-08:
Sketch:

* `POST /runs/{runId}/commands` gains two new query parameters:
    * `waitUntilComplete`: `true` or `false`, defaulting to `false`.
    * `timeout`: Integer or float seconds from 0 to ~30, defaulting to ~30.
    * We could probably combine these if we really wanted to, but having two explicit parameters feels safer and more explicit to me, especially with our parsing framework's historically confusing handling of `null`, `undefined`, and missing fields.
* If `waitUntilComplete` is `false`, return immediately, as it currently does.
* If `waitUntilComplete` is `true`, return when the command’s status becomes either succeeded or failed, *or* when `timeout` is reached, whichever comes first. The timeout will start from when the command is inserted into the Protocol Engine.
* Status code is still 200 if the request times out. It's the client's job to inspect the result command's `status`.
    * If the result command’s status is "succeeded", the command completed successfully within the timeout.
    *  If the result command’s status is "failed", it failed within the timeout. Note that if the run is stopped while commands are still queued, those commands are marked "failed".
    *  If the result command’s status is "queued" or "running", the timeout passed.

mcous, 2022-02-08:
> timeout: Integer or float seconds from 0 to ~30, defaulting to ~30.

- Milliseconds could be a little more JS friendly
- Dots in query parameters freak me out for no good technical reason 🙃 

> Status code is still 200 if the request times out

- I think this is the right move, and we need to be careful in the docs accordingly
- Status code should be `201 Created`, though