Opentrons/opentrons #9414: API: pick_up_tip() presses parameter only uses default

alexjoel42, 2022-02-09:
# Overview
When trying to set the amount of presses in tip pickup, the pipette still only uses default configurations.  

# Steps to reproduce

1. Be on Software version 4.7.0 or later 
2. Use the "presses" parameter in the protocol ( for example p20.pick_up_tip( TIP.[‘A1’], presses=0) )
3. In this case, it won't hover over the tip rack. It will just 

```Python3
p20.pick_up_tip( TIP.[‘A1’].top(z=10), presses=0) 

```

# Current behavior
Will press tip pickup the default amount. 

# Expected behavior

Follows the correct amount of presses identified by the API documentation.

# Theory about what happened
## In 4.7.0 we see 

1. [Hardware controller](https://github.com/Opentrons/opentrons/blob/40fcfd9ee628dccace4a8e118d0039030ebd400c/api/src/opentrons/hardware_control/api.py#L956-L990) no longer have code to use different press quantities. 
2. API [is similar too](https://github.com/Opentrons/opentrons/blob/edge/api/src/opentrons/protocol_api/instrument_context.py#L642-L700)

## 4.6.2
We see the [hardware controller](https://github.com/Opentrons/opentrons/blob/hotfix_4.6.2/api/src/opentrons/hardware_control/api.py#L1610-L1630) have conditional logic for different quantities of presses

alexjoel42, 2022-02-09:
Looks like this is following a general request for clarification in our API documentation [about what the pick_up_tip()](https://docs.opentrons.com/v2/new_protocol_api.html#opentrons.protocol_api.contexts.InstrumentContext.pick_up_tip) command can do https://github.com/Opentrons/opentrons/issues/8505

TamarZanzouri, 2022-07-19:
Tested this on a robot. Passing argument presses=3 to pick_up_tip() works as expected. Confirming with @alexjoel42 what work needs to be done for this ticket.

TamarZanzouri, 2022-07-20:
confirmed with @sfoster1, this is a bug and we need to allow presses=0

PR #11200 (2022-07-22): fix(PAPIv2): allow presses=0 to hover over tip when calling pick_up_tip
# Overview

closes #9414.
allow argument `presses=0` when calling pick_up_tip() in APIv2

# Changelog

- changed condition in plan_check_pick_up_tip to check for None instead of not
- Added a new test suite to test pipette_handler.py

# Review requests

- [x] tested on a robot.
- [ ] I really want to add a test for the hardware api layer to make sure _move is being called once when supplied presses=0 but its turning into a rabbit whole. any suggestions? 
- [ ] are the tests sufficient? 

# Risk assessment

Very small chang [... cut]
closed 2022-07-22 by TamarZanzouri