Opentrons/opentrons #9416: feat(engine): analyze ctx.load_module during protocol analysis

mcous, 2022-02-09:
## Overview

As a user uploading a PAPIv3 protocol for analysis, I would like the analysis to report module usage.

## Acceptance criteria

- [ ] A PAPIv3 protocol uploaded via `POST /protocols` reports back module loads and usage in its analysis

## Implementation notes

At the moment, a simulating `ProtocolRunner` + `ProtocolEngine` will fail on `loadModule` commands due to the lack of attached hardware to the simulator. Broadly, I see our two best options as:

1. Use or augment the `HardwareAPI` in simulator mode to get module hardware simulators
    - Something like the logic in `HardwareAPI.find_modules`
    - Do not modify the ProtocolEngine
    - This is similar to how PE currently analyzes `loadPipette` commands
2. Add a config to the protocol engine to "virtualize" modules
    - Something like what we do in the ProtocolEngine to analyze `ctx.pause` in PAPIv3

I'm quite partial to exploring option (2), but option (1) may be a better place to start

SyntaxColoring, 2022-02-10:
Hey team! Please [add your planning poker estimate](https://app.zenhub.com/workspaces/opentrons-cpx-5e1f6ed55b295a3f514f2283/issues/Opentrons/opentrons/9416?planning-poker) with ZenHub @mcous @sanni-t @y3rsh

mcous, 2022-02-14:
Taking option number 2, and leaving the HW API out of it for now. We'll might have to come back in and fold in the hardware simulator when we get to module commands, but for now option 2 should be sufficient to hit the acceptance criteria

PR #9460 (2022-02-14): refactor(engine): allow module loads to be analyzed in PAPIv3
## Overview

This PR implements module load analysis in PAPIv3. Closes #9416.

## Changelog

- Add a `use_virtual_modules` config to the `ProtocolEngine`
- Set `use_virtual_modules=True` in `create_simulating_runner`
- Return a fake load result from `EquipmentHandler.load_module` if `use_virtual_modules` is set

## Review requests

- [ ] Code and tests, including smoke test, make sense
- [ ] Module loads come back from `POST /protocols` with PAPIv3 protocols

## Risk assessment

Very low
commit 52f176c3 (2022-02-14): refactor(engine): allow module loads to be analyzed in PAPIv3 (#9460)

Closes #9416
closed 2022-02-14 by mcous