Opentrons/opentrons #9458: PAPIv3: get and set rail lights

mcous, 2022-02-14:
## Overview

As a PAPIv3 protocol author, I would like to turn the rail lights on and off, as well as retrieve their current status.

## Acceptance criteria

PAPIv3 authors may:

- [ ] Use `context.set_rail_lights` to set the rail lights on or off
- [ ] Use the `context.rail_lights_on` property to retrieve lights on/off state

## Implementation notes

- Setting the rails lights will require a rail lights PE command that does not yet exist
- Retrieve the rail light state presents an interesting problem. Do we:
    1. Mirror the hardware state in the engine state, so that the context can retrieve this value from the engine state
    2. Provide a facade to hardware state from the engine, so the context can "directly" read the state from the hardware

I think option (2) is less fraught from a source of truth standpoint, so should be our default implementation path. Option (1) isn't awful, and has a _little_ bit of precedence in existing engine state, but maybe only due to cruft rather than intentional design.

y3rsh, 2022-02-23:
How to make a light show against the http API
```python
"""Light Show"""
import time
import random
import requests

session = requests.Session()
session.headers = {"Opentrons-Version":"2"}
lights = "http://localhost:31950/robot/lights"

for n in range(0, 10):
    session.post(lights,json={"on": True})
    time.sleep(random.random())
    session.post(lights,json={"on": False})
    time.sleep(random.random())
```
⍰ How does this connect to hardware controller?  To turn the lights on? [Hardware Control](https://github.com/Opentrons/opentrons/blob/e549c8c20205f7575700ed4df0adae18990305a9/api/src/opentrons/hardware_control/api.py#L312)
⍰ This does not go through any engine?

y3rsh, 2022-02-23:
```gherkin
Feature: set_rail_lights and rail_lights_on

Given I use PAPIv3
When I context. set_rail_lights(true)
Then context.rail_lights_on() == true

Given I use PAPIv3
When I context. set_rail_lights(false)
Then context.rail_lights_on() == false

Given I use PAPIv3
When I context. set_rail_lights(false)
And I context. set_rail_lights(true)
Then context.rail_lights_on() == true

Given I use PAPIv3
When I context. set_rail_lights(true)
And I context. set_rail_lights(false)
Then context.rail_lights_on() == false

Given I use PAPIv3
When I context. set_rail_lights(true)
And I context. set_rail_lights(true)
Then context.rail_lights_on() == true

Given I use PAPIv3
When I context. set_rail_lights(false)
And I context. set_rail_lights(false)
Then context.rail_lights_on() == false

Given I use PAPIv3
When I context. set_rail_lights(true)
And I pause
And during the pause, I use the app or HTTP API to turn OFF the lights 
Then after the pause, context.rail_lights_on() == false

Given I use PAPIv3
When I context. set_rail_lights(false)
And I pause
And during the pause, I use the app or HTTP API to turn ON the lights 
Then after the pause, context.rail_lights_on() == true
```

y3rsh, 2022-02-23:
```python
import time
import random

metadata = {
    "protocolName": "v2 tests",
    "author": "Josh McVey <josh.mcvey@opentrons.com>",
    "description": "v2 exercise",
    "apiLevel": "3.0",  # works on "apiLevel": "2.9"
}

binary = [True, False]

def run(ctx):
    for n in range(0, 10):
        # ctx.set_rail_lights(random.choice(binary))
        ctx.set_rail_lights(True)
        pause_seconds = random.randrange(1, 5)
        # ctx.pause(f"Pause for {pause_seconds} seconds")
        time.sleep(pause_seconds)
        # ctx.resume()
        ctx.set_rail_lights(False)
```

mcous, 2022-04-01:
"Get rail lights state" moved to #9863

closed 2022-04-01 by mcous