Opentrons/opentrons #9471: PAPIv3: support comment command

mcous, 2022-02-15:
## Overview

As a PAPIv3 author, I want to add comments to my protocol that will appear in the app's run log, so that I can inspect the protocol run and give instructions to the operator.

## Acceptance criteria

A PAPIv3 author may:

- [ ] Issue a comment via `context.comment`

## Implementation notes

- This command does not exist in the engine, but obviously, its implementation can be a no-op
- For some inspiration, keep #356 in mind. We're not addressing it now, but we have the space to do so in #8359

alexjoel42, 2022-07-14:
@mcous , does this deserve a separate ticket for PAPIv2? I noticed when running protocols with comments (on software versions 5.0.0 --> 6.0.2 haven't tested beyond), it doesn't go to the api.log file. 

This might be related to this bug https://github.com/Opentrons/opentrons/issues/11131 
```python3

from opentrons import protocol_api

metadata = {'apiLevel': '2.12'}

def run(protocol: protocol_api.ProtocolContext):
    plate = protocol.load_labware('corning_96_wellplate_360ul_flat', 1)
    tiprack_1 = protocol.load_labware('opentrons_96_tiprack_300ul', 2)
    p300 = protocol.load_instrument('p300_multi_gen2', 'left', tip_racks=[tiprack_1])
    protocol.pause(msg='please stop')
    protocol.comment("example 1")
    protocol.comment("example 2")

```

[serial.log](https://github.com/Opentrons/opentrons/files/9112490/serial.log)
[server.log](https://github.com/Opentrons/opentrons/files/9112491/server.log)
[api.log](https://github.com/Opentrons/opentrons/files/9112492/api.log)

mcous, 2022-07-27:
@alexjoel42 no, this is unrelated feature ticket about aligning the Python Protocol API with the newer protocol execution system introduced with JSON v6 protocols.

As a rule, we make no guarantees about what does or does not go into `api.log`. The logs are a debugging tool. If you wants to find out what commands did or did not happen in a run, as of v6, you should use the runs HTTP API:

- `GET /runs` - get all the runs a robot has performed (capped at the last 20 runs)
- `GET /runs/:run_id` - get the details of a specific run (you can get the ID from `GET /runs`)
- `GET /runs/:run_id/commands` - get the commands list of a given run

closed 2023-06-03 by mcous