Opentrons/opentrons #9479: bug: Robot status displayed as running while LPC is exited and robot homing

nusrat813, 2022-02-15:
# Overview

When a user completes LPC the robot is homing, while the robot is homing user clicks proceed to run and is now in the run tab. The run tab displays the status of the robot as running and only lets user pause run instead of start run. Once the homing is complete the user then sees the button change to start run

# Steps to reproduce

# Current behavior
[image]

# Expected behavior

[image]

b-cooper, 2022-05-13:
confirm this is not a problem with smoketest before release candidate is tagged

mikecardarelli, 2022-06-03:
@hartm9 to check if this is still happening

hartm9, 2022-06-03:
This behavior is still happening - I grabbed a screenshot of the app while the robot was homing after LPC: 
[image]

After clicking the Pause Run button, the homing routine paused, and clicking Resume Run caused the protocol to begin.

b-cooper, 2022-06-07:
This is no longer reproducible in edge, as of recent changes to the robot-server that make sure the run status is not set to "running" until after LPC is finished and the run has been initiated.

closed 2022-06-07 by b-cooper