Opentrons/opentrons #9503: bug: Opentrons app lists wrong protocol steps

silvtal, 2022-02-18:
# Overview

The in-app simulation lists wrong steps, even impossible ones, which makes it harder to follow the protocol while running. This issue happened before 5.0.0. as well. This doesn’t happen when simulating with opentrons_simulate on the terminal.

# Steps to reproduce
- Use 364-well plates and a multi-channel pipette
- **Use custom 200 uL tips instead of the default 300 uL ones**
- Include a loop such as the following in the Python protocol

```
for r in range(3): # 1 reservoir well --> 6 columns of the plate ; 3 wells total                                            
    d = r*6
    p300multi.distribute(100,    
                         reservoir.wells()[r+first_r_well].bottom(mm), 
                         plate.columns()[0+d : 6+d],                   
                         blow_out=False, 
                         disposal_volume=0,    
                         new_tip="never")
```

# Current behavior

The app states that the pipette is aspirating 300 uL (which is impossible with the 200 uL tips), distributing 100+100+100 and aspirating again. However, what it really is doing is aspirate 200 uL and fill one of the 364 columns before aspirating again (distributes 100 uL in the A row + distributes 100 uL in the B row).

[image]

Before running, the whole list of steps is wrong. When it starts running, once the step has been performed it gets "corrected" like this

[image]

As you can see, it's still wrong.

# Expected behavior

The app should list the correct steps

mcous, 2022-02-18:
@silvtal does this behavior change if you set the robots "use slower analysis" advanced setting?

I appreciate the detailed issue and screenshots. Are you able to post a minimal, but complete protocol that exhibits this behavior to make it a little easier on us to test and reproduce?

silvtal, 2022-02-23:
Sorry for not providing a minimum example. I didn't have access to the robot this week. Thanks for looking into this!

mcous, 2022-02-23:
No worries @silvtal, your report gave us a lot of helpful information to investigate this issue! I can also confirm that turning on the "use slower analysis" setting should resolve the incorrect UI problem, for the time being, at the expense of taking longer to initially load the protocol

mcous, 2022-02-23:
Adding an even more simple repro protocol:

```python
from opentrons import types

metadata = {
    "protocolName": "Tip Volume Testing",
    "apiLevel": "2.12",
}

def run(ctx):
    tr = ctx.load_labware("opentrons_96_filtertiprack_200ul", 1)
    p300 = ctx.load_instrument("p300_single", types.Mount.RIGHT, [tr])
    reservoir = ctx.load_labware("nest_1_reservoir_195ml", 3)

    p300.pick_up_tip()
    p300.aspirate(location=reservoir.wells()[0].bottom(1))
    p300.drop_tip()
```

This one is particularly bad. The analysis fails, claiming the protocol will fail, even though it will run perfectly well.

- Analysis: 1x aspirate 300 uL, fails because 300 uL exceeds tip volume
- Actual run: 1x aspirate 200 uL, succeeds

PR #9550 (2022-03-29): fix(api): working volume is not set in fast simulation
# Overview

Our fast simulation was never setting the `working_volume` when picking up a tip. This led to bad transfer planning.

closes #9503 

# Changelog

- Added unit test to highlight the error
- Fix the error

# Review requests

I tested using the protocol provided in the bug description

# Risk assessment

Moderate?
commit dfbf2377 (2022-03-29): fix(api): working volume is not set in fast simulation (#9550)

* pipette dict parity between fast and slow sims

* set working volume when picking up tip.

* get_available_volume matches slow sim calculation.

* update available_volume when current_volume changes. update ready_to_aspirate. clamp working_volume using pipette config max.

* tip length

* with tip tests work.

* flow rate

* speed and flow rate

closes #9503