Opentrons/opentrons #9515: bug: Actual magnet height does not match UI when controlling a Magnetic Module through the Opentrons App

SyntaxColoring, 2022-02-18:
# Overview

When controlling a Magnetic Module through the Opentrons App, magnet heights will be wrong.

The actual height of the magnets will be systematically shorter than the value that the user entered, and the value that the Opentrons App displays. I believe this happens for both GEN1 and GEN2 modules, even when you account for the difference in units between the two.

For example, setting a GEN2 Magnetic Module's magnet height to 10 mm through the app will actually set it to 7.5 mm.

Also, attempting to set a height below the bottom of the labware, which the UI implies is possible, does nothing.

# Steps to reproduce

I tested this on v5.0.0, but I'd also expect it to be a problem on v4.7.0.

1. Connect and turn on a Temperature Module GEN2. The magnets should home. Measured with calipers, they should be 2.5 mm below the ledge on which the labware sits.
    * Note: I'm getting the value 2.5 from [this Python constant](https://github.com/Opentrons/opentrons/blob/9fadafce7cd575ae3db08aca7e5257dd6c186ddc/api/src/opentrons/hardware_control/modules/magdeck.py#L20), which matches what I see with calipers ± 0.2 mm. This value does *not* match what the Opentrons App says is the disengage point, -4 mm, [defined here](https://github.com/Opentrons/opentrons/blob/9fadafce7cd575ae3db08aca7e5257dd6c186ddc/shared-data/js/constants.ts#L88-L97). I don't know where this discrepancy comes from. Possibly, this discrepancy is a separate bug.
2. SSH into the robot and tail the logs with `journalctl --follow --no-pager`.
3. In the Opentrons App, go to **Modules** and select **Engage.**
   [image]

4. Enter 0 mm and select **Set Height.**
    * **Expected behavior:** The magnets should move up so that their tops are level with the ledge on which the labware sits.
        * I expect this because of the "GEN2 Height Ranges" box that the Opentrons App shows.
    * **Actual behavior:** The magnets do not move.
5. Enter -1 mm and select **Set Height** again.
    * **Expected behavior:** The magnets should move so that their tops are 1 mm below the ledge on which the labware sits.
        * See step (4) for why I expect this.
    * **Actual behavior:** The magnets do not move. Also, in the robot logs, `ValueError: Invalid engage height for magneticModuleV2: -1 mm. Must be 0 - 25 mm` shows up.
6. Enter 10 mm and select **Set Height** again.
    * **Expected behavior:** The magnets should move up so that their tops are 10 mm above the ledge on which the labware sits.
        * See step (4) for why I expect this.
    * **Actual behavior:** The magnets *do* move up. But, measured with calipers, they're only 7.5 mm above the ledge, which is 2.5 mm shorter than expected. The app displays the current height as 10 mm.

# Technical details

The **Set Height** button sends this HTTP payload to `POST /modules/{serialNumber}`:

```json
{"command_type":"engage","args":[10]}
```

(In this example, I entered 10 mm in the height field in the app.)

This is not quite correct. It's [equivalent to](https://github.com/Opentrons/opentrons/blob/9fadafce7cd575ae3db08aca7e5257dd6c186ddc/robot-server/robot_server/service/legacy/routers/modules.py#L129) a Python call to [`MagDeck.engage(10)`](https://github.com/Opentrons/opentrons/blob/9fadafce7cd575ae3db08aca7e5257dd6c186ddc/api/src/opentrons/hardware_control/modules/magdeck.py#L95), which means *10 mm from the home position,* not 10 mm from the bottom of the labware.

So, *with the current HTTP API,* the Opentrons App either needs to:

* Adjust its UI to tell the user that it's measuring from the module's home position, not from the bottom of the labware.
* Do math to convert distances from the bottom of the labware to distances from the module's home position. The app will need to take care to properly handle the difference in units between GEN1 and GEN2 Magnetic Modules. Also, we will need to figure out why the app and server appear to be using different height ranges, as mentioned in the steps to reproduce.

With the next iteration of the HTTP API for module control, which we're currently thinking through, we might have better options.

skyewindsor, 2022-05-31:
Nusrat reproduced this today. She entered a 9mm engage height into the app, and the module card showed an engage height of 11.5mm

[image] 

[image]

mcous, 2022-06-13:
Closed by #10724. App should now reflect back the exact values the user sends in the command. The following magnetic module issues are still open:

- #10736
- #9529

Of the two, I think #10736 should definitely be in 6.0.0, and 9529 is a bigger issue that needs careful consideration to avoid breaking existing protocols, so will be fixed in 6.1.0 at the earliest

PR #10724 (2022-06-13): fix(robot-server): calculate magnetic module gen 1 height 
## Overview

Closes #9515.

Engage command implementation adjusts Gen1 magdeck height from base.
Need to do the opposite when getting the height.

## Changelog

- Subtract home-to-base offset from HW reported magnetic module height
- Half reported height for GEN1 magnetic modules to account for half-mm units from HW API

## Review requests

- Code and tests look good
- Works on hardware*

### Known issues

*On GEN1 magnetic modules, the engage height that the module moves to will be about 1.5mm too low; see #9585. In other words, if yo [... cut]
commit c231d1b8 (2022-06-13): fix(robot-server): calculate magnetic module gen 1 height  (#10724)

Closes #9515
closed 2022-06-13 by mcous